Cloud-Automation/node-modbus #4: client code error

sandeepbs39, 2015-05-28:
C:\node server>node mod.js
module.js:338
    throw err;
          ^
Error: Cannot find module 'xmlhttprequest'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (C:\node server\node_modules\socket.io\node_modules\so
cket.io-client\node_modules\engine.io-client\lib\transports\index.js:5:22)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)

C:\node server>

Cloud-Automation, 2015-05-28:
There is no xmlhttprequest in this code base. What are you trying to do?

sandeepbs39, 2015-05-28:
i'm trying to run that example code

sandeepbs39, 2015-05-28:
C:\node server>node mod.js
C:\node server\node_modules\socket.io\node_modules\socket.io-client\node_modules
\engine.io-client\lib\transports\polling.js:23
  var xhr = new XMLHttpRequest({ xdomain: false });
            ^
TypeError: object is not a function
    at C:\node server\node_modules\socket.io\node_modules\socket.io-client\node_
modules\engine.io-client\lib\transports\polling.js:23:13
    at Object.<anonymous> (C:\node server\node_modules\socket.io\node_modules\so
cket.io-client\node_modules\engine.io-client\lib\transports\polling.js:25:3)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (C:\node server\node_modules\socket.io\node_modules\so
cket.io-client\node_modules\engine.io-client\lib\transports\polling-xhr.js:6:15)

    at Module._compile (module.js:460:26)

C:\node server>

Cloud-Automation, 2015-05-28:
We'll first of all this is a node.js module. So you cannot run it inside the browser. Second of all this example is still not very useful and I don't think you have a problem with this module.

closed 2015-05-28 by Cloud-Automation