Cloud-Automation/node-modbus #13: Example missing token and method not implement

Mirdrack, 2016-01-20:
Hello:
First of all I want to tank you for develop this module, its so clear an easy to use

On the example in the read me are miss the close tokens for the function **"});"** but the real question is if the the **"readHoldingRegister"** function is implemented because I take a look in the library code and i could not find it. 
```
client.readHoldingRegister(0, 10, function (reps, err) {
    // resp will look like { fc: 3, byteCount: 20, register: [ values 0 - 10 ] }
    console.log(err, resp); 
```

This gets my attention because you put it in example and when I run the example get an error with that problem.

Thanks a lot in advance

Cloud-Automation, 2016-01-20:
What error do you get? And Version do you use?

Mirdrack, 2016-01-21:
Im really sorry, looks like I install it before the **readHoldingRegister** were included on this commit 9e852713cf6764efb1cf670485774f3f7974ae87
I was receiving this error
```
client.readHoldingRegister(0, 10, function (reps, err) {
       ^
TypeError: undefined is not a function
```
But was for and old version of the code, all becasuse I didnt realize the function was used on the client example before were implement it

Thanks again, Ill keep tracking this great project

closed 2016-01-21 by Cloud-Automation