Cloud-Automation/node-modbus #15: Working over Modbus RTU/ASCII

bladerunner2020, 2016-01-28:
I have been using jsModbus module for more than year in my smart home installation. It works perfectly and I would like to thank you very much for great job.

I am working on another project where I need to access modbus devices over Modbus RTU/ASCII. I haver RS-485 to USB adaptor and I curious is it possible to use/adapt your module to use it with serial port?

Cloud-Automation, 2016-01-28:
Since I separate the MBAP Header from the PDU in the client it shouldn't be that hard to implement. Unfortunatly I do not own a modbus slave device, so I have no idea how to test this. 

Are there more people who like to see this as a feature in 0.6.0?

bladerunner2020, 2016-01-28:
I can test. And with some help I could try to implement this...

bladerunner2020, 2016-01-29:
Great! Have you any idea when 0.6.0 could be released?

Cloud-Automation, 2016-01-29:
Not yet, sorry.

BauchBeinePoe, 2016-02-24:
Have a look at the v1.0.0 alpha pull request.

BauchBeinePoe, 2016-03-11:
Hey @bladerunner2020 can you test the v1.0.0 modbus over serial functions. Unfortunatly I can't.

bladerunner2020, 2016-03-14:
Will try to check this week.

closed 2016-03-24 by BauchBeinePoe

bladerunner2020, 2016-04-02:
Finally I got some time for testing but got stuck with serial port. It failed to install on Win7-64. If you could help me with binaries for serial port. 

`Error: Cannot find module '/Users/Bladerunner/Documents/Node JS Projects/playwithnode/node_modules/serialport/build/Release/node-v47-darwin-x64/serialport.node'`

BauchBeinePoe, 2016-04-04:
You better handle this on the serialports github page https://github.com/voodootikigod/node-serialport.