Cloud-Automation/node-modbus #19: Memory leaks

bladerunner2020, 2016-02-21:
It seems that there is a memory leak in jsmodbus module.
I have 16 modbus devices connected via TCP. Each device has 8 registers.
I wrote a simple test that reads all registers of all devices every 5 seconds.
For 40 minutes the heap grew from 36 to 436 Mb. See the picture attached.

Test application is attached.

[test_modubs.txt](https://github.com/Cloud-Automation/node-modbus/files/139783/test_modubs.txt)

[image]

BauchBeinePoe, 2016-02-24:
As I see it you are executing a async function in a loop, this is bad practice. Try to make the next readHoldingRegister request when the first one is finished. Have a look [here](http://stackoverflow.com/questions/4288759/asynchronous-for-cycle-in-javascript) on howto make async loops.

This could lead to the big heap!

Another thing is that you do not to want make a single request for every single word. Try making it so `client.readHoldingRegister(0, reg_addresses.length, cb)`. You must eventually make multiple calls if the length from the reg_addresses is bigger than the maximum allowed in the modbus protocol.

This is the part i am a talking about.

    for (var i = 0; i < reg_addresses.length; i++) {
        client.readHoldingRegister(reg_addresses[i], 1, function (resp, err) {
            if (err) {
                console.log("Error reading register %s from device ID %s. Closing client.", this.register, this.clientID);
                console.log(err);
                modbus_clients[clientID].close();
                return;
            }

            // console.log ("Register %s from client #%s read. Data = %s", this.register, this.clientID, resp.register[0]);
        }.bind({register: i, clientID : clientID}));
    }

Can you confirm that?

bladerunner2020, 2016-02-27:
Hi Stefan! 

Thank you very much for your remark. I rewrote the code using async lib. See attached.
It doesn't solve the issue with memory leaks, but it was slowed down. I did the test and the help grew to 314 Mb from 24 Mb during 3 hours.

What concerns your suggestion to read multiple registers at once... It's a good advice as well. I tried to use readDiscreteInput and this function didn't work on my mobile device. I will re-write the code again and do some testing.

[image]

[app.txt](https://github.com/Cloud-Automation/node-modbus/files/149313/app.txt)

bladerunner2020, 2016-02-27:
I rewrote the code again to read 8 registers at once instead of each separately. I run the test and the memory leaks were gone! That I don't understand - what was wrong with the previous code?

[image]

[app.txt](https://github.com/Cloud-Automation/node-modbus/files/149509/app.txt)

BauchBeinePoe, 2016-02-28:
I would say that, since you have read the single registers separatly the 1000 ms cycle was to long. I mean the 16 modbus requests took longer than 1000ms. Try measuring how long a modbus request takes and I bet the eight or 16 requests tooks longer than 1s.

The problem here is that, as i assume, some request haven't been finished by the time the next interval function is called, the heap grows because of unfinished modbus requests and every seconds there will be some more requests that haven't been finished.

closed 2016-02-28 by BauchBeinePoe

bladerunner2020, 2016-02-28:
It's quite reasonable. Thanks again for assistance.

Sisco21, 2016-04-20:
Thanks,

it's working great now. GC do the job. Just one question, i'm reading Holding registers and i'm limited with the max address which is 40000. This is normal? On our automat we could address more than this limit.

BauchBeinePoe, 2016-05-20:
Can you open up a new issue for that?