Cloud-Automation/node-modbus #22: createTCPServer() async call caused error

NNM11, 2016-03-10:
There is a async issue with the jsmodbus.js  createTCPServer() call (line 91).

            var server = serialServerModule.create(
            tcpServer,
            handler.Server.RequestHandler,
            handler.Server.ResponseHandler);
	    // async - server is unknown as the cb is executed before the create() call completed.
            cb(null, server);

Occasionally, the code will fail if the server create() call takes too long.

BauchBeinePoe, 2016-03-11:
Can you confirm that branch v0.5.2 fixes that?

NNM11, 2016-03-11:
The v0.5.2 branch did not fix the problem.

BauchBeinePoe, 2016-03-14:
Ah, I didn't saw that there is a full error description including code. I checked the simpleServer.js example and it works if you set the port to 502. Check the err variable before assigning handler.

NNM11, 2016-03-15:
The sample code with error checking will avoid this problem when port 502 is blocked by default.

     // create Modbus TCP Server
     modbus.createTCPServer(8888, '127.0.0.1', function (err, modbusServer) {
         if (err) { 
                   console.log(err);
                   process.exit(1);
          }
          // addHandler
          modbusServer.addHandler(4, rirHandler);
          modbusServer.addHandler(5, writeCoilHandler);
    });

closed 2016-03-15 by NNM11