Cloud-Automation/node-modbus #25: Register number doubled in the TCP server

profelec, 2016-03-26:
Hi,

I think that the register number is doubled compared to the real number:
my server :
```
var serverModbus = modbus.server.tcp.complete({ 'logEnabled'    : false, 
                                                'port'          : CONF_portAutomate, 
                                                'hostname'      : CONF_ipAutomate,
                                                'responseDelay' : 10 });
 
serverModbus.on('readHoldingRegistersRequest', function (byteStart, quantity) {    
    console.log('readHoldingRegistersRequest (' + byteStart + ', ' + quantity + ')');   
});

serverModbus.on('preWriteSingleRegisterRequest', function (byteAddress, value) {   
    console.log('write single register (' + byteAddress + ', ' + value + ')');
});
```

my client :
```
// Ecrit de type holding - registre n° 16
client.writeSingleRegister(16, parseInt(data, 2)).then(function (resp) {}).fail();

// Registre n°20 - lecture de 1 registre - Fonction n°3 (holding)
client.readHoldingRegisters(20, 1).then(function (resp) {}).fail();
```

And the result:
```
readHoldingRegistersRequest (40, 1)
write single register (32, 2)

```
instead of:
```
readHoldingRegistersRequest (20, 1)
write single register (16, 2)
```

BauchBeinePoe, 2016-03-26:
That is right. The registers in the server are ArrayBuffer and byte addressed. The function codes are word (16bit) addressed. I am not sure if the modbus specification says anything about that. In order to be able to write to address 1 and 2 (for example the holding register)  without overlapping I doubled the address values.

profelec, 2016-03-26:
Ok, thanks.

closed 2016-03-26 by BauchBeinePoe