Cloud-Automation/node-modbus #26: Modbus TCp

mikegreenpower, 2016-03-27:
Hi,

can you show to me how to use your Modbus TCC module on web site?

BauchBeinePoe, 2016-03-27:
What exactly are you planing to do?

mikegreenpower, 2016-03-27:
Hi,
i have a web application for data monitoring ( data from meter) . In this application i want to add a section in wich i can send modbus TCP command to my instruments for setting.

BauchBeinePoe, 2016-03-28:
Have a look at the modbus specification at Modbus.org and get familiar with the protocol. It is a simple protocol and once you know some function Codes you will be able to use this Module with ease.

jakehawkes, 2016-03-29:
It is probably worth mentioning that you will need a server to run the node Modbus module in, which will be where the commands are sent from. The commands cannot be sent from the webpage. That is, they cannot be sent from the webpage running on a client computer. The webpage will instead result in the Modbus command being sent from the computer serving the webpage.

closed 2016-03-31 by BauchBeinePoe