Cloud-Automation/node-modbus #27: How complete is the serial code?

grahamreeds, 2016-03-30:
The reason I ask I am looking into adding automation to a project I am working on. The application already has a node interface. All values are logged to SQLite. I need to decide between:

1. Changing the values directly based on user input via node-modbus (or similar).
2. Writing the values to socket.io and having a small C app change the values via libmodbus.
3. Read the logged values in a small C app and change the values via libmodbus.

1 is the preferred option as it reduces the points of failure. It does requires a solid ModbusRTU implementation and the text for the project is Modbus TCP (no mention of Serial) and the serial-client code was only added a month ago.

The questions I have are:

- Is the serial code ready for industrial automation apps?
- Is there much code planned for it?
- Will there be a specific RTU module?

BauchBeinePoe, 2016-03-31:
You can use a untested serial client/server but it is not ready for production since I do not have a serial modbus device.

Try this and give me some feedback so I can complete the serial part.

    var client = require('jsmodbus').client.serial.complete;

    client({ 'baudrate' : 9600, 'portName' : 'usb0' });

From here on it is the usual client API. Have a look at src/modbus-serial-client.js. The module uses the serialport module.

workinghoward, 2016-05-19:
Hey again, I have some test, and seems like the serial-port you create is undefined, I think there's some typo on your "modbus-serial-client.js".
`
serialport = new SerialPort(this.portName, {
                baudRate : this.baudrate,
                parity : 'even'
            });`
It should be baudRate : this.baudRate

and I modified this one and I still can not get the data from modbus simulator on my pc.

BauchBeinePoe, 2016-05-20:
Since the serial part is not ready, can you tell me what setup you are using and can you use a com port sniffer to see what is happening.

reejinbouk, 2016-06-22:
hi, how to connect to serial port to /dev/ttyUSB0 , 9600 , 8n1?
thank you

BauchBeinePoe, 2016-06-27:
By using the `portName` setting. 

For example:
`client({baudrate:9600, portName: 'usb0' });`

BauchBeinePoe, 2016-07-06:
Updated to serialport 3.0.0.

reejinbouk, 2016-07-14:
but how to setup parity and stop - data bits?

BauchBeinePoe, 2016-07-18:
I will forward those to the serialport. Just give me some time :-)

iurly, 2016-08-09:
Hi,
I'm trying to use the serial implementation. My ultimate goal would be to actually use node-RED, for which a wrapper node is available (node-red-contrib-modbus, though without a serial implementation yet), based on jsmodbus.
I'm currently testing this on a local linux PC, without the actual serial device, using the following setup:

1) create a pair of "virtual" serial ports:
$ socat -d -d pty,raw,echo=0 pty,raw,echo=0

2016/08/06 16:34:16 socat[26147] N PTY is /dev/pts/4
2016/08/06 16:34:16 socat[26147] N PTY is /dev/pts/5
2016/08/06 16:34:16 socat[26147] N starting data transfer loop with FDs [5,5] and [7,7]

2) Use a **patched** pymodslave to simulate a serial device: https://sourceforge.net/projects/pymodslave/files/

where I changed ModSlaveSim.py so use use /dev/pts:
            _modServer = modbus_rtu.RtuServer(serial.Serial(port="/dev/pts/"+str(int(args[1])),

This way I found that modbus-serial-client.js actually has a broken CRC implementation, which looks more like a checksum:

            for (var i = 0; i < buf.length; i += 1) {
                crc = (buf.readUInt8(i) + crc) % 0xFFFF; 
            }

Shall I open a separate issue?

BauchBeinePoe, 2016-08-09:
Yes please. And feel free to fork the Projce since I am quiet busy right now. Great work so far. I will try to reproduce that.

BauchBeinePoe, 2016-08-24:
Since @iurly is very destined to get on with the serial implementation. The 1.1-dev branch is the place to look.

k30r88, 2016-08-30:
Hi, I am wondering if this modbus serial implementation here is Modbus RTU over serial ?

SuperJojo2001, 2016-09-26:
Dear Developers, there is a Modbus Simulation Software available for windows at http://www.plcsimulator.org/
This software simulates the full range of 254 Mobus RTU slaves and it supports Modbus TCP as well. It is perfect to test the package against a real PC base Modbus slave. Download the installer and the two registry files to register the app.

closed 2017-07-28 by stefanpoeter