Cloud-Automation/node-modbus #28: All tests in modbus-server-core.test.js fails...

grahamreeds, 2016-03-30:
...but all tests in modbus-client-core.test.js pass.

I am running Ubuntu 14.04 LTS with node v0.10.25 and npm 1.3.10.

I am thinking the version of node might be too low but as this is an industrial unit we can't change it.

```
  Modbus Server Core Tests.
    Exception Tests.
      1) should answer with 0x8x status code due to missing handler.
    Read Coils Tests.
      2) should handle a read coils request just fine.
      3) should handle a read coils request with a start address outside the address space.
      4) should handle a read coils request with a quantity value outside the address space.
    Read Discrete Inputs Tests.
      5) should handle a read discrete inputs request just fine.
      6) should handle a read discrete inputs request with a start address outside the address space.
      7) should handle a read discrete inputs request with a quantity value outside the address space.
    Read Holding Registers Tests.
      8) should handle a read holding registers request just fine.
      9) should handle a read holding registers request with a start address outside the address space.
      10) should handle a read holding registers request with a quantity value outside the address space.
    Read Input Registers Tests.
      11) should handle a read input registers request just fine.
      12) should handle a read input registers request with a start address outside the address space.
      13) should handle a read input registers request with a quantity value outside the address space.
    Write Single Coil Tests.
      14) should handle a write single coil request just fine.
      15) should handle a write single coil request with a start address outside the address space.
      16) should handle a write single coil request with a another value than 0x0000 (false) and 0xff00 (true).
    Write Single Register Tests.
      17) should handle a write single register request just fine.
      18) should handle a write single register request with a start address outside the address space.
    Write Multiple Coils Tests.
      19) should handle a write multiple coils request just fine.
      20) should handle a write multiple coils request with a start address outside the address space.
      21) should handle a write multiple coils request with a start and quantity outside the address space.
    Write Multiple Registers Tests.
      22) should handle a write multiple registers request just fine.
      23) should handle a write multiple registers request with a start address outside the address space.
      24) should handle a write multiple registers request with a start and quantity outside the address space.
      25) should handle a write multiple registers request with a quantity greater 0x007b.

  0 passing (324ms)
  25 failing

  1) Modbus Server Core Tests. Exception Tests. should answer with 0x8x status code due to missing handler.:
     TypeError: Object � has no method 'compare'
      at resp (test/modbus-server-core.test.js:22:39)
      at Object.<anonymous> (src/modbus-server-core.js:51:17)
      at Context.<anonymous> (test/modbus-server-core.test.js:28:18)

  2) Modbus Server Core Tests. Read Coils Tests. should handle a read coils request just fine.:
     Uncaught TypeError: Object  has no method 'compare'
      at resp (test/modbus-server-core.test.js:49:39)
      at Object.<anonymous> (src/modbus-server-core.js:59:17)
      at Object.<anonymous> (src/handler/server/ReadCoils.js:69:17)

  3) Modbus Server Core Tests. Read Coils Tests. should handle a read coils request with a start address outside the address space.:
     Uncaught TypeError: Object � has no method 'compare'
      at resp (test/modbus-server-core.test.js:67:39)
      at Object.<anonymous> (src/modbus-server-core.js:59:17)
      at Object.<anonymous> (src/handler/server/ReadCoils.js:44:21)

  4) Modbus Server Core Tests. Read Coils Tests. should handle a read coils request with a quantity value outside the address space.:
     Uncaught TypeError: Object � has no method 'compare'
      at resp (test/modbus-server-core.test.js:85:39)
      at Object.<anonymous> (src/modbus-server-core.js:59:17)
      at Object.<anonymous> (src/handler/server/ReadCoils.js:44:21)

[... 105 lines cut ...]
  22) Modbus Server Core Tests. Write Multiple Registers Tests. should handle a write multiple registers request just fine.:
     Uncaught TypeError: Object  has no method 'compare'
      at resp (test/modbus-server-core.test.js:498:39)
      at Object.<anonymous> (src/modbus-server-core.js:59:17)
      at Object.<anonymous> (src/handler/server/WriteMultipleRegisters.js:70:17)

  23) Modbus Server Core Tests. Write Multiple Registers Tests. should handle a write multiple registers request with a start address outside the address space.:
     Uncaught TypeError: Object � has no method 'compare'
      at resp (test/modbus-server-core.test.js:519:39)
      at Object.<anonymous> (src/modbus-server-core.js:59:17)
      at Object.<anonymous> (src/handler/server/WriteMultipleRegisters.js:52:21)

  24) Modbus Server Core Tests. Write Multiple Registers Tests. should handle a write multiple registers request with a start and quantity outside the address space.:
     Uncaught TypeError: Object � has no method 'compare'
      at resp (test/modbus-server-core.test.js:537:39)
      at Object.<anonymous> (src/modbus-server-core.js:59:17)
      at Object.<anonymous> (src/handler/server/WriteMultipleRegisters.js:52:21)

  25) Modbus Server Core Tests. Write Multiple Registers Tests. should handle a write multiple registers request with a quantity greater 0x007b.:
     Uncaught TypeError: Object � has no method 'compare'
      at resp (test/modbus-server-core.test.js:555:39)
      at Object.<anonymous> (src/modbus-server-core.js:59:17)
      at Object.<anonymous> (src/handler/server/WriteMultipleRegisters.js:41:21)

```

grahamreeds, 2016-03-31:
Just upgraded my development installation to v4.4.1 and all tests now pass.  The question remains, why do the client-core tests pass but the server-core fail despite looking very similar?

(My background is C++ not JS.)

BauchBeinePoe, 2016-03-31:
> 1) Modbus Server Core Tests. Exception Tests. should answer with 0x8x status code due to missing handler.:
     TypeError: Object �� has no method 'compare'
      at resp (test/modbus-server-core.test.js:22:39)
      at Object.<anonymous> (src/modbus-server-core.js:51:17)
      at Context.<anonymous> (test/modbus-server-core.test.js:28:18)

The compare method is not used in the client tests. Must be something with the Put Module and the node version i guess.

closed 2016-03-31 by BauchBeinePoe