Cloud-Automation/node-modbus #29: Memory leak ?

Sisco21, 2016-04-19:
Hello, and congratulation for your code. I've just a question about a supposed memory leak.

I try to read holding register all 2000 ms with the code below (i read 50 objects)

> var readHoldingRegister = function (objectId,startAddress, objectLength,writeAdresseOffset,writeObjectLength) {
> 
> 	var startAddressObject = parseInt(startAddress) + parseInt(objectId)*parseInt(objectLength);
> 	var writeAddressForObject = parseInt(startAddress) + parseInt(writeAdresseOffset) + (parseInt(writeObjectLength) * parseInt(objectId));
> 
>     intv = setInterval(function () {
> 		if(isModbusConnected)
> 		{
> 			loggerProdModBus.debug("___");
> 			loggerProdModBus.debug("Call read holding register %s",objectId);
> 			var pRead = clientProxyModBus.readHoldingRegisters(startAddressObject, objectLength);
> 			var pWrite = clientProxyModBus.readHoldingRegisters(writeAddressForObject, writeObjectLength)
> 			
> 			Promise.all([pRead, pWrite]).then(function(values) {
> 				if(values.length === 2)
> 					parserModBus.ParseModBusRegister(objectId,values[0].register,startAddressObject,values[1].register,writeAddressForObject);
> 				else
> 					loggerProdModBus.error('Promise all not returning expected result. Result :',values)
> 
> 			}).catch(function(err)
> 			{
> 				loggerProdModBus.error('Error with ModBus Server and reading readHoldingRegister',err)
> 			});
> 			
> 		}
>     }, pause);
> 
> };

But after few minutes, the memory used reach 512 Mb and stop. There is a solution to clear memory once the job is done?

Thanks

François

bladerunner2020, 2016-04-19:
Please check #19. I had the issue and it was solved when I rewrote the code using async approach.

BauchBeinePoe, 2016-04-19:
The code you wrote seems fine. How often do you call this function? Making two Modbus calls every two seconds doesn't seem to be so memory intensive.

Can you provide some more Code?

BauchBeinePoe, 2016-04-19:
If you want to periodically update Registers try the jsmodbus-tools Library in npm. It does what you describe without the memory Problem. It currently supports Holding Register only but I can extend it to Coils and Input Register.

Sisco21, 2016-04-20:
Thanks

It's working great now. Holding registers seem to be limited to 40000 (Max address)?

It's right? On our automat, we could go higher than this limit.

Best regards

François

BauchBeinePoe, 2016-04-20:
If I remember correctly that is the maximum value for an address determined in the Modbus specs.

closed 2016-04-20 by BauchBeinePoe