Cloud-Automation/node-modbus #30: How can I set the reconnect time interval?

workinghoward, 2016-04-26:
I have been working on a project by this awesome jsmodbus, but when I enable the AutoReconnect, 
program will do the reconnect high-frequency, even raise the CPU utility to 100 %.
So I want to know is there any function I can call to set the reconnect time?
Thanks

BauchBeinePoe, 2016-04-26:
Normally establishing a connection through TCP/IP takes some time to fail. What I am curious about is why is the connection not established so fast. Can you post some code and maybe some debug messages ( set the logEnable to true and the logLevel to debug).

workinghoward, 2016-05-17:
Sorry about my late reply.
I was trying to complete a node-red module with reconnect feature by jsmodbus.
I had seen  one but without reconnect feature.
I try to create some situation like un-plug the cable or something, like server off-power for like an hour.
so If the reconnect is keep trying too fast, it seems like too much work for a gateway.
So I'd like to know is there any way to slow the reconnect time?

Thanks

BauchBeinePoe, 2016-05-18:
One way would be to set the `autoReconnect` attribute to `false` and listen for the `close` event and manually reconnect.

````

client.on('close', function () {

    setTimeout(client.connect, 5000);

});

````

workinghoward, 2016-05-19:
Thanks for the help, but I have a another issue in serial module
Di I need to open new issue for this?

closed 2016-05-19 by workinghoward