Cloud-Automation/node-modbus #31: onSocketError doesnt emit "error"

mainstreetmark, 2016-05-26:
My devices which are offline are not getting the "error" event.

It looks like you need to add `this.emit('error')` in `onSocketError` in modbus-tcp-client.js maybe?

BauchBeinePoe, 2016-06-08:
Sorry for the late reply.

I am on it.

BauchBeinePoe, 2016-06-08:
Can you verify?

mainstreetmark, 2016-06-08:
Yes, the callback works, though I don't get a numeric code in the results.

    code: 'ECONNREFUSED',
    errno: 'ECONNREFUSED',

Otherwise, consider it closed.  :)

mainstreetmark, 2016-06-08:
Actually, that callback may be firing hundreds of times.  I log when i try to connect, and I log on('error') and i get lots and lots of errors per connect, within a second.  My timeout is set to 1000ms, so every 1000ms, i get a hundred console.logs.

BauchBeinePoe, 2016-06-08:
Can you post some example code.

mainstreetmark, 2016-06-08:
```
var modbus = require('jsmodbus');

// create a modbus client
var client = modbus.client.tcp.complete({
	'host': "10.90.1.199",
	'port': 8888,
	'autoReconnect': true,
	'timeout': 1000,
	'unitId': 0
});

client.connect();

function DoRead() {
	client.readInputRegisters(0, 9).then(function(resp) {
			console.log("input", resp.register);
		})
		.fail("input fail", console.log)
		.done(setTimeout(DoRead, 1000));
	client.readCoils(0, 9).then(function(resp) {
			console.log("coils", resp.coils);
		})
		.fail("coil fail", console.log);
}

client.on('connect', function() {
	DoRead();
});

client.on('error', function(err) {
	console.log("ERROR", err);
});

```

The ("ERROR", err) line appears hundreds of times the instant I stop the master (which is just running in another terminal window on the same machine)

BauchBeinePoe, 2016-06-08:
The code looks solid. 

I just checked: the reconnect mechanism does not use a timeout. The timeout is related to the client requests. So whenever a connection is closed due to a connection loss or an error the reconnect sets in. 

Let me check tomorrow if I can quickly use the timeout value for the reconnect or event make it a new setting.

closed 2016-06-27 by BauchBeinePoe