Cloud-Automation/node-modbus #33: Use serialport 3.x

krewenki, 2016-06-21:
When using serialport 2.x (the current version in package.json), I am getting multiple deprecation warnings : 

```
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.

==== JS stack trace =========================================

Security context: 0x1ac4aa5c9fa9 <JS Object>#0#
    1: .node [module.js:568] [pc=0xaa688c9eae4] (this=0x21bbad9cc241 <an Object with map 0x9feb7d17e49>#1#,module=0x2d49ae1766d9 <a Module with map 0x9feb7d18421>#2#,filename=0x2d49ae1766a1 <String[87]: /Users/warren/Sites/telegauge/TG4/node_modules/serialport/build/Release/serialport.node>)
    2: load [module.js:456] [pc=0xaa688c39152] (this=0x2d49ae1766d9 <a Module with map 0x9feb7d18421>#2#,filename=0x2d49ae1766a1 <String[87]: /Users/warren/Sites/telegauge/TG4/node_modules/serialport/build/Release/serialport.node>)
    3: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0xaa688c38c7d] (this=0x1ac4aa504189 <undefined>,module=0x2d49ae1766d9 <a Module with map 0x9feb7d18421>#2#,filename=0x2d49ae1766a1 <String[87]: /Users/warren/Sites/telegauge/TG4/node_modules/serialport/build/Release/serialport.node>)
    4: _load [module.js:407] [pc=0xaa688c348c2] (this=0x21bbad9cc0f9 <JS Function Module (SharedFunctionInfo 0x21bbad922dd1)>#3#,request=0x2d49ae175791 <String[87]: /Users/warren/Sites/telegauge/TG4/node_modules/serialport/build/Release/serialport.node>,parent=0x2d49ae166259 <a Module with map 0x9feb7d18421>#4#,isMain=0x1ac4aa504299 <false>)
992ae] (this=0x2d49ae15fde9 <an Object with map 0x3876ef007ac1>#6#,exports=0x2d49ae15fde9 <an Object with map 0x3876ef007ac1>#6#,require=0x2d49ae161479 <JS Function require (SharedFunctionInfo 0x21bbad953b41)>#7#,module=0x2d49ae15fd99 <a Module with map 0x9feb7d18421>#8#,__filename=0x2d49ae15fd61 <String[73]: /Users/warren/Sites/telegauge/TG4/node_modules/serialport/lib/bindings.js>,__dirname=0x2d49ae161411 <String[61]: /Users/warren/Sites/telegauge/TG4/node_modules/serialport/lib>)
    9: _compile [module.js:541] [pc=0xaa688c41c24] (this=0x2d49ae15fd99 <a Module with map 0x9feb7d18421>#8#,content=0x2d49ae160a91 <String[784]\: 'use strict';\n\nvar bindings = require('bindings')('serialport.node');\nvar listUnix = require('./list-unix');\n\nvar linux = pr
==== C stack trace ===============================

 1: v8::Template::Set(v8::Local<v8::Name>, v8::Local<v8::Data>, v8::PropertyAttribute)
 2: SerialportPoller::Init(v8::Local<v8::Object>)
 3: init
 4: node::DLOpen(v8::FunctionCallbackInfo<v8::Value> const&)
 5: v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&))
 6: v8::internal::MaybeHandle<v8::internal::Object> v8::internal::(anonymous namespace)::HandleApiCallHelper<false>(v8::internal::Isolate*, v8::internal::(anonymous namespace)::BuiltinArguments<(v8::internal::BuiltinExtraArguments)1>)
 7: v8::internal::Builtin_HandleApiCall(int, v8::internal::Object**, v8::internal::Isolate*)
 8: 0xaa688b0961b
```

I updated my local copy to use serialport 3, and the warnings are gone.  I didn't run any tests, so maybe it just breaks everything, but there's errors on the horizon anyway, using the current module.

BauchBeinePoe, 2016-06-27:
Thanks for the hint. I will make an update.

BauchBeinePoe, 2016-07-06:
So far so good! :-)

closed 2016-07-06 by BauchBeinePoe