Cloud-Automation/node-modbus #39: Examples not working

reejinbouk, 2016-08-19:
Hello
trying examples for serial -  not working
whats wrong?

var ModbusClient  = require('jsmodbus'),
    client = ModbusClient('/dev/ttyUSB0', 9600);
client.on('connect', function () {
    client.writeSingleCoil(4, true).then(function (resp) {
        console.log(resp);
    }).fail(function (err) {
        console.log(err);
    }).done(function () {
        client.close();
    });
});
client.on('error', function (err) {
    console.log(err);
});

got this:
TypeError: ModbusClient is not a function

when i change second to this:
    client = ModbusClient.createSerialClient('/dev/ttyUSB0', 9600);
got this:
TypeError: ModbusClient.createSerialClient is not a function

please update examples?

reejinbouk, 2016-08-20:
Second question - how to setup ID of modbus device (serialport)

BauchBeinePoe, 2016-08-22:
You are using it wrong. Try the following : 

    var ModbusClient = require('jsmodbus'),
          client = ModbusClient.client.serial.complete({ 'portName' : '/dev/ttyUSB0',  'baudRate' : 9600 });

I will update the examples.

Can you confirm it works? But be aware, the serial code is not done it.

mjasnikovs, 2016-08-23:
```modbus connected
c:\hub\modbus\boot.js:14
        client.readInputRegister(0, 13)
               ^

TypeError: client.readInputRegister is not a function
    at Object.client.on (c:\hub\modbus\boot.js:14:9)
    at Object.emit (c:\hub\modbus\node_modules\stampit-event-bus\src\stampit-event-bus.js:20:38)
    at Object.<anonymous> (c:\hub\modbus\node_modules\jsmodbus\src\modbus-tcp-client.js:57:18)
    at emitNone (events.js:91:20)
    at Socket.emit (events.js:185:7)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1068:10)```

Same here, example is not working.

BauchBeinePoe, 2016-08-23:
It should be  `client.readHoldingRegisterS`.

mjasnikovs, 2016-08-23:
Pleas update readme "client.readDiscreteInputS" and "client.readInputRegisterS"

Thank you.

closed 2016-08-23 by BauchBeinePoe