Cloud-Automation/node-modbus #43: 'index out of bound' when using writing multiple coils

bam-ftw, 2016-09-08:
There seems to be an issue with writing multiple coils. I have tried it using jsmodbus client as well as another application.
Server always crashes with jsmodbus client, no matter number of coils.
With QModMaster application server crashes when number of coils exceeds 7.

Example code: [https://github.com/bam-ftw/jsmodbus-write-multiple-coils-error](https://github.com/bam-ftw/jsmodbus-write-multiple-coils-error)

BauchBeinePoe, 2016-09-17:
I can confirm the bug with the current version. It seems to be a client problem since it works with pull request #45. Handling more than 7 bits still causes a server crash, so this one is on the server side.

BauchBeinePoe, 2016-09-17:
Can you check branch fc15_server_bug and confirm that the bug is fixed?

bam-ftw, 2016-09-18:
- Writing multiple coils no longer causes crashes. 
- Writing up to 8 coils works correctly.
- Writing more than 8 coils causes invalid values to be stored in server coils buffer. I have updated my repository to show this issue.

Example code: [https://github.com/bam-ftw/jsmodbus-write-multiple-coils-error](https://github.com/bam-ftw/jsmodbus-write-multiple-coils-error)

BauchBeinePoe, 2016-10-15:
Latest version 1.2.3 fixes this issue. Thanks for reporting @bam-ftw.

closed 2016-10-15 by BauchBeinePoe