Cloud-Automation/node-modbus #46: v1.0 not working on raspberry pi 2

profelec, 2016-09-15:
Hi, 

I use this great library on my raspberry, but with the passage on the next version, i can't install this.
The errors messages : 
`
> serialport@2.1.2 install /home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/./serialport.h:7:0,
                 from ../src/serialport.cpp:1:
../node_modules/nan/nan.h:328:47: error: 'REPLACE_INVALID_UTF8' is not a member of 'v8::String'
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
serialport.target.mk:84: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.13-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.4.13-v7+
node-pre-gyp ERR! command "node" "/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.26
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/NodeJS/N3-PorteAuto/v4/node_modules/jsmodbus/node_modules/serialport/build/Release' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! serialport@2.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.1.2 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.13-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "update"
npm ERR! cwd /home/pi/NodeJS/N3-PorteAuto/v4
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/NodeJS/N3-PorteAuto/v4/npm-debug.log
npm ERR! not ok code 0

`

profelec, 2016-09-15:
serialport is not supported on arm v7 : 
https://www.npmjs.com/package/serialport#platform-support

profelec, 2016-09-15:
```
curl -sL https://deb.nodesource.com/setup_4.x | sudo -E bash -
sudo apt-get install -y nodes
sudo apt-get install -y build-essential
sudo npm install -g node-gyp
sudo npm install --unsafe-perm serial port
```
fixed my problem

BauchBeinePoe, 2016-09-17:
Great work @profelec. I mark this as fixed.

closed 2016-09-17 by BauchBeinePoe