Cloud-Automation/node-modbus #49: Issue through node-red-contrib-modbus

SuperJojo2001, 2016-09-24:
Hello, I am a user of the node-red-contrib-modus. The developer told me that in his sources he's using the node-modbus and whenever I have a problem with modbus I should submit an issue here.

I have the following problem: 
First I have today that I am using the node-red-contrib-modbus in Modbus RTU serial mode.

When I am issuing a simple "Read Holding register" at addresss 0 with 1 register length  I see the node pushing the stream 0x01 0x03 0x00 0x00 0x00 0x01 0x00 0x05 to the serial line. (I have made it visible through an analyser). The point is that the checksum at the end is sent as 0x00 0x05, but this is wrong and 
must be 0x84 0x0A. 
Watchching the source source code I see the following code sequence, which tells me that the checksum is build by a simple addition...but this is not right.

for (var i = 0; i < buf.length; i += 1) {
                crc = (buf.readUInt8(i) + crc) % 0xFFFF; 
}

The checksum of Modbus RTU has to be build in accordance to the Modbus RTU spec with a real cyclic redundancy check and polynomial. See the spec here 
http://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf

SuperJojo2001, 2016-09-26:
Have seen that the npm crc package already includes the calculation of the Modbus Serial crc:
https://www.npmjs.com/package/crc

SuperJojo2001, 2016-09-26:
We added the following lines to correct the CRC.
First include the CRC library
var crc = require('crc');

then instead of the data stream addition we put use this code sequence

crc16 = 0;
crc16= crc.crc16modbus(buf);
pkt = pkt.word16le(crc16).buffer();

Now the correct CRC is appended and we can communicate to a modbus slave first time.

BauchBeinePoe, 2016-10-08:
Hey @SuperJojo2001,

i've seen that you files also issues #50 and #51. Why don't you make all of these a pull request and i'll have a look.

Great work so far.

SuperJojo2001, 2016-10-08:
Done

SuperJojo2001, 2016-10-08:
Created the pull request #56

closed 2016-10-08 by SuperJojo2001