Cloud-Automation/node-modbus #54: Serial doesn't work in my case

biancode, 2016-10-03:
Hi,

you have some handling of PDU for TCP but not for Serial.
Result of testing ModbusRTU and ASCII: serial doesn't work in my case

The handling with endian isn't ready:
there is just ...BE for BIG-Endian in use and there is now way to use the ..LE for Little-Endian

Serial connection type is just RTU? could be ASCII or RTU

ASCII needs a SerialPort parser with line ending '\r\n' or 0x0d 0x0a or (10,13)LE / (13,10)BE

Serial doesn't work for PDU's with UnitID and/or split of PDU:

DEBUG : received data PDU: {"type":"Buffer","data":[1,3,2,0]}
DEBUG : received data PDU: {"type":"Buffer","data":[54,56,82]}

Serial-Source:  [1,3,2,0,54,56,82]  (LITTLE)
UnitID: 1
FC: 3
16BitBlocksToRead: 2
16BitBlock1-readLE: payload -> 54
16BitBlock2-readLE: CRC

[1,3,2,0] and [54,56,82] should be: [1,3,2,0,54,56,82]
UnitID (1), FC (1), ByteCount, Value (2), CRC (2)

If the serial raise often an exception on read:

> 2 Oct 22:38:37 - [red] Uncaught Exception:
2 Oct 22:38:37 - RangeError: index out of range
    at checkOffset (buffer.js:799:11)
    at Buffer.readUInt16LE (buffer.js:845:5)
    at Object.<anonymous> (node_modules/node-modbus/node-modbus/handler/client/ReadHoldingRegisters.js:93:48)
    at Object.<anonymous> (node_modules/node-modbus/modbus-client-core.js:152:13)
    at Object.emit (node_modules/stampit-event-bus/src/stampit-event-bus.js:20:38)
    at Object.<anonymous> (node_modules/node-modbus/modbus-serial-client.js:103:18)
    at emitOne (events.js:90:13)
    at SerialPort.emit (events.js:182:7)
    at SerialPort.module.exports.raw (node_modules/serialport/lib/parsers.js:7:13)
    at SerialPort._emitData (node_modules/serialport/lib/serialport.js:310:18)
    at SerialPort.<anonymous> (node_modules/serialport/lib/serialport.js:290:14)
    at SerialPort.<anonymous> (node_modules/serialport/lib/serialport.js:303:7)
    at FSReqWrap.wrapper [as oncomplete] (fs.js:614:17)

BauchBeinePoe, 2016-10-08:
Sounds to me like a feature request to switch litte/big. I will see to it in the next version.

SuperJojo2001, 2016-10-08:
As far as I see the implementation today just Modbus RTU is supported only. ASCII I see nowhere coded. Next to 0x0d 0x0a at the end of an ASCII frame it needs also a semikolon (':', 0x3A) at the start of the frame.
We should get Modbus RTU working fine first, then we can add ASCII as well. After issuing pull request #56 Modbus RTU works fine with me now. @biancode Wondering how you receive PDU data correctly without #56?

biancode, 2016-10-08:
@SuperJojo2001 I fixed some in my repo branch https://github.com/biancode/node-modbus-1, but didn't do any pull request for now. Testing isn't finished.

closed 2016-12-16 by biancode