Cloud-Automation/node-modbus #57: Is it possible to get the support of ModbusRTU over TCP

GermanBluefox, 2016-10-14:
There are some Modbus RTU <=> TCP gateways in the world.
They just take serial RTU frame and send them via TCP (no MBAP header, but with CRC).
Is it possible to support it?

tschombe, 2016-10-18:
I would also like to see this feature, a good explanation about it can be found here:
http://www.simplymodbus.ca/TCP.htm

GermanBluefox, 2016-10-19:
I implemented this feature and can create the PR for this.

https://github.com/ioBroker/ioBroker.modbus/blob/master/lib/modbus-tcp-rtu-client.js

BauchBeinePoe, 2016-10-24:
Please, do so and I have a look.

BauchBeinePoe, 2016-11-21:
See Pull Request #81 .

closed 2016-11-21 by BauchBeinePoe

theopract, 2019-03-20:
Hi Stefan,

I can see you closed the issue but looks like you never merged the pull request. 
Can we expect integration of "Modbus RTU over TCP" functionality in the near future?

stefanpoeter, 2019-03-20:
It shouldn't be that hard to implement but i myself don't have the need for Modbus RTU over TCP. I am always happy to look into new Feature Requests.

theopract, 2019-03-20:
Ok, thank you.