Cloud-Automation/node-modbus #62: Can the two's complement be decoding incorrectly?

angieskazka, 2016-11-09:
Hi! i'm getting data from this device and negative temperatures show up in node.js application as 65413-something. I can fix it by manually post-processing the inputs like this:
```
  var decode = function(x){
    if(x>1000) x = x - 65536; //substract Math.pow(2,16)
    return x;
  }
```
but i wonder if node-modbus already supposed to interpret this correectly on reading the registers. the "two's complement" and so on...

Here is my device i'm reading from 
http://www.comet-america.com/products/temperature-humidity-pressure-transmitters-and-regulators/web-sensor-p8641-with-poe-four-channels-remote-thermometer-hygrometer/reg-P8641#downloadd

in the instruction manual [here](http://www.comet-america.com/userfiles/file/manuals-english/transmitters/ie-snc-P86xx.pdf?time=1478686246&FixForIE=.pdf) page 22 there is a register map: 

[image]

BauchBeinePoe, 2016-11-21:
If I understand this correct, you are wondering how you can get the two-complement interpretation of your temperature? Since @psorowka made a pull request, you are able to read the result from the modbus request out of a buffer. From that you can just `readUInt16` your temperature. Have a look [here](https://nodejs.org/api/buffer.html) for more help.

closed 2016-11-21 by BauchBeinePoe

angieskazka, 2016-11-30:
thanks @BauchBeinePoe! it helped (of course it did)