Cloud-Automation/node-modbus #90: Newer node Version required

valfis86, 2016-11-24:
Hi,

your package.json states that a node Version >= 0.6 is required. It seems like your a using some features only present in newer versions of nodejs.  Yesterday we ran into the following error message:

"TypeError: Buffer.allocUnsafe is not a function". (jsmodbus/src/handler/client/ReadHoldingRegisters.js:49)

Updateing our node Version from 4.x.x to 6.9.1 solved the issue. 

Please update the package.json file with the correct minimum nodejs version.

Kind regards,

Valentin

BauchBeinePoe, 2016-11-24:
You mean Version >= 6 right?

psorowka, 2016-11-24:
@valfis86 can you tell which Node Version you really used before? I made the change to `allocUnsafe` and checked the latest Node 4 docs before: https://nodejs.org/dist/latest-v4.x/docs/api/buffer.html and thus assumed that would be a safe decision

GermanBluefox, 2016-11-24:
allocUnsafe is of course .... bad. My platform ioBroker is based actually on 4.x 
allocUnsafe appeared in 5.10.
That means, modbus will not run on ioBroker. And I do not have possibility to go to 5, 6, 7 just now because not all packages are run now in newer versions.

GermanBluefox, 2016-11-24:
This one https://nodejs.org/dist/latest-v4.x/docs/api/buffer.html could be a solution. I was irritated by
[image]

BauchBeinePoe, 2016-11-24:
Hey, that is fun, here is my offer:
[image]

from the same html page.

psorowka, 2016-11-24:
shame on me :( @GermanBluefox please see #91 we will rollback the breaking changes and I strongly propose adding Travis CI so this is prevented in the future.

BauchBeinePoe, 2016-11-24:
That is interessting, the [latest doc page](https://nodejs.org/docs/latest/api/buffer.html#buffer_class_method_buffer_allocunsafe_size) says **added in v5.10.0** and the one you (refer)[https://nodejs.org/dist/latest-v4.x/docs/api/buffer.html] to says **Added in v4.5.0**.

psorowka, 2016-11-24:
Yes I figure that was more or less something that happened in parallel. Node 5 development was started when Node 4 was in LTS so they added features simultaneously to the 4 and 5 branches. The whole modernized `Buffer` handling (that favors `Buffer.from` over `new Buffer`) obviously landed in `Node 4.5`.  I am already so used to this that I didn't realize that it was so recent :-D

@GermanBluefox is updating to >= Node 4.5 an option for you?

valfis86, 2016-11-24:
@psorowka We used node version 4.2.1

GermanBluefox, 2016-11-24:
>@GermanBluefox is updating to >= Node 4.5 an option for you?

Of course.

My users start to disturbing me with an error **then(...).fail(...)** .
@BauchBeinePoe will you contact the npm support? Or should I modify my code?

BauchBeinePoe, 2016-11-24:
I already contacted the support. No reply yet.

BauchBeinePoe, 2016-11-28:
I'll close this one. There is a new 2.0.0 Version online and the old 1.2.6 version is actually the 1.2.5 version. So everything is fine.

closed 2016-11-28 by BauchBeinePoe