Cloud-Automation/node-modbus #91: Changing Version 1.2.6 to 2.0.0

BauchBeinePoe, 2016-11-24:
As mentioned in #89 and #90 the latest changes have an effect on the API and on the backwards compatibility that I did not noticed. I would suggest to rollback the latest version (1.2.6) and file a new 2.0.0 version instead. @psorowka Any thoughts on that?

psorowka, 2016-11-24:
uuuuuuhh there are still people using the old node versions? :-D I didn't realize that the current version is packaged for node >= 0.6. Sorry for that. I was aware that the Buffer Handling changed but reading the Node4 docs (https://nodejs.org/dist/latest-v4.x/docs/api/buffer.html) I assumed I wouldn't break anything. Didn't test it, though.

I would propose that we add a Travis CI hook so we don't run into such breaking changes in the future.

I think it is a good idea to rollback the latest version and start with 2.0.0. What would you think about requiring Node 6 starting the 2.0.0? This is the current LTS and would allow to use the latest ES6 features, like `let` and `const` scoping and some more..

GermanBluefox, 2016-11-24:
> Node 6 starting the 2.0.0

No
4.6 is LTS too.

If you have 20 dependencies you can migrate to new version. If you have 2000...

BauchBeinePoe, 2016-11-24:
Travis CI sounds like a good idea. I am gonna do some digging, haven't used it. Didn't I implement tests for the Function Code APIs. Why didn't they report any errors on the new Promise API?

I am ok with 2.0.0. I will then unpublish the 1.2.6 on npmjs and remove the tag in the repository. I'll leave the master in tact so. I will also make a new pull request for features in the 2.0.0 release. Maybe @GermanBluefox is speeding up with the test for pull request #85 so this can be in there too.

Also applying standard javascript style is a must then.

Requiring node 6 for the 2.0.0 release!

BauchBeinePoe, 2016-11-24:
@GermanBluefox But 4.6 doesn't have new ES6 features.

GermanBluefox, 2016-11-24:
>But 4.6 doesn't have new ES6 features.

It worked in 4.x and why suddenly the package requires ES6? Just for "can"?

GermanBluefox, 2016-11-24:
> unpublish the 1.2.6

You can't. Only if not older than 30 minutes. But there is still an option to write to support. I tested it, it works.

BauchBeinePoe, 2016-11-24:
```` npm ERR! "cannot unpublish a version that is >24 hours old" : jsmodbus ````

About the version. I am also fine with 4.x.

BauchBeinePoe, 2016-11-24:
See Pull Request #92

psorowka, 2016-11-24:
Ok, so the 2.x will be the new master and you open a 1.2.x branch for the compatibility, right? Do you also plan to backport features to the old branch (hassle :) ) or will you essentially freeze it? 

In any case we should add a `.travis.yml` with the older relevant versions to that other branch for the sake of completeness right?

BauchBeinePoe, 2016-11-24:
@psorowka A good point. I'll skip the part with the backport features ;-) I will create a new branch for 1.2.x and I think bug fixing this branch should be no problem (for a certain time). No new features thou. :-) I'all also add a travis.yml.

BauchBeinePoe, 2016-11-24:
Branch v1.2.x is the new branch.

BauchBeinePoe, 2016-11-28:
Version 2.0.0 is released.

psorowka, 2016-11-29:
npm release is missing :)

BauchBeinePoe, 2016-11-30:
Now it is :-)

closed 2016-11-30 by BauchBeinePoe