Cloud-Automation/node-modbus #96: Promise.defer is deprecated

SystemParadox, 2016-11-30:
Node 6 prints lots of warnings to the console:

```
(node:7143) Warning: Promise.defer is deprecated and will be removed in a future version. Use new Promise instead.
```

Any chance this could be updated?

Thanks.

BauchBeinePoe, 2016-11-30:
I'll fix this on the next update (2.0.1). Could be a few days though.

psorowka, 2016-11-30:
ah.. how did you produce these warnings? I didn't see any of them when switching to that.

Actually I knew it would be deprecated but I figured the chances that @BauchBeinePoe accepts the PR is higher when I don't completely change everything :-D

@BauchBeinePoe I can file a PR for that later today. It is my fault anyway :)

BauchBeinePoe, 2016-11-30:
@psorowka It's alright. I already fixed it. I'll publish a PR later.

BauchBeinePoe, 2016-12-01:
@SystemParadox can you check if your warnings are still there with the v2.0.1-Release Pull Request?

SystemParadox, 2016-12-02:
Warnings are gone. Looks good to me. Thanks :)

BauchBeinePoe, 2016-12-05:
See #98

closed 2016-12-05 by BauchBeinePoe