Cloud-Automation/node-modbus #97: Throw error if no error listeners are added

SystemParadox, 2016-11-30:
When using node's EventEmitter, if no `error` listeners are added, emitting an error should result in the error being thrown.

This does not seem to happen with `jsmodbus`. Trying to connect to a device which does not exist will silently fail unless I do a `client.on('error', ...)`.

I spent a while looking through the code but I am completely confused by this `stampit` thing. I can't find where `emit` is defined, nor can I find any reference to `EventEmitter`.

BauchBeinePoe, 2016-11-30:
> When using node's EventEmitter, if no error listeners are added, emitting an error should result in the error being thrown.

Is this a feature request?

We are using the `stampit-event-bus` module to handle events. You can read about stamps [here](https://www.npmjs.com/package/stampit).

SystemParadox, 2016-12-02:
It's a bug report or feature request depending on your viewpoint.

You've implemented an EventEmitter-like interface. I just think that it should follow the same semantics as Node's EventEmitter so as to avoid surprises. It surprised me at least.

BauchBeinePoe, 2016-12-02:
Throwing Exceptions in the event of an error seems like a good idea. However this is a change in the API and might break Implementations since an uncatched exception can terminate the whole application. 

I'd rather make a note on the README that there are no exception triggered in the event of an error. @psorowka and @SystemParadox what are your thoughts on that?

psorowka, 2016-12-02:
I think the recent 2.0.0 tag is a good chance to introduce such behavior. I agree to @SystemParadox  that this is the usual behavior

SystemParadox, 2016-12-02:
My preference would be to standardise on the EventEmitter behaviour. However, you could argue that since errors are not necessarily fatal (due to automatically reconnecting) it might be valid not to throw by default.

If your preference is to keep the current behaviour I'll go with that. Maybe something like:

"Unlike Node's EventEmitter, errors will not be thrown if there are no error listeners added".

To be honest it should probably be noted in the README either way what the default error behaviour is.

Thanks.

SystemParadox, 2016-12-02:
Is it worth considering the addition of an `error.fatal` flag, if there are some errors that are fatal and some that will still allow a reconnect?

BauchBeinePoe, 2016-12-02:
Let's see. The most errors are related to requests that are made and those are handled in promisses. The only errors I can think of are distrupted connections. How would you want to catch exception errors anyway?

````
var client = modbus.client.tcp.complete(...);

try {

    client.connect();

} catch (e) {
...
}

````

I mean this is an async event so handling this in a try catch block doesn't make much sense, does it?

What if a connection fails after it is established? Throwing an exception in this case means having a big try and catch block.

SystemParadox, 2016-12-02:
As you say this is all async so there is no `try...catch`. I was thinking this was all be part of the error event still, so I could do something like this:

```
client.on('error', function (err) {
    if (err.fatal) {
        // thrown asynchronously
        throw err;
    } else {
        console.error(err.message);
    }
});
```

If no listener is added at all, it would either be thrown asynchrously (like Node's EventEmitter), or do nothing (the current behaviour).

However, it sounds like no errors are really fatal, so perhaps the current behaviour actually makes more sense.

BauchBeinePoe, 2016-12-02:
You are right. There are no fatal errors and connection errors should be taken in consideration when using communication modules. Also handling connection errors with a `process.on('uncaughtException'...)` is not really a good way to go here, the way I see it this would be the same as `client.on('error', ...)`.

lfernando-silva, 2016-12-02:
I had a smilar problem before, because I was depending of the connection refused error to follow the system procedures. At that time, I made a TCP ping before proceed the procedures, that I described here in the  [https://github.com/Cloud-Automation/node-modbus/issues/36](https://github.com/Cloud-Automation/node-modbus/issues/36) issue. If you have a better solution, I'm very interested!

closed 2016-12-05 by BauchBeinePoe