Cloud-Automation/node-modbus #99: Creating three buffers x 1024 bytes in server mode contains undefined data after init

SuperJojo2001, 2016-12-01:
Allocating a buffer using the .buffer method as used in modbus-server-core.js the buffer returned contains uncertain data.

This is why I recommend to use the method buf.fill(0) function as stated here https://nodejs.org/api/buffer.html#buffer_new_buffer_array after the allocation.

On my hardware using node-red-contrib-modbus which is using modbus-server-core.js as dependency I see weired data after a reset. I have to write all data myself first before I get it all to zero. The problem is that my application might be to late in doing so and a polling Modbus Client might be faster polling than I can fill the buffer all with zeros within my application.

BauchBeinePoe, 2016-12-01:
What .buffer method do you mean @SuperJojo2001? The buffers used in the modbus-server-core are created with `new Buffer(...)` which is deprecated. `Buffer.alloc(size, fill)` should be used instead. I'll file this for v2.0.1.

You can create buffers for the server by using the initial buffers like so

````
var server = modbus.server.tcp.complete({ 
    holding: Buffer.alloc(1024, 0),
    coils: Buffer.alloc(1024, 0),
    input: Buffer.alloc(1024, 0),
    port : 8888 
});
````

SuperJojo2001, 2016-12-01:
Yes you are absolutely right. Buffers are created with _Buffer(1024)_ function currently and not with ._buffer_ ... my fault and a typo. 
And yes this method is deprecated, as I have seen it in the node.js description. 
If you gonna use the Buffer.alloc() function instead in future then I am fine.
Thx

BauchBeinePoe, 2016-12-02:
See PR #98

closed 2016-12-02 by BauchBeinePoe