Cloud-Automation/node-modbus #103: Server example doesn't work - this.on is not a function

robogeek, 2016-12-05:
I'm new to modbus and am trying this module to get up to speed.  However, I can't get to the starting gate.  I tried the Server example from the README -- and I get this output:

```
$ node --version
v6.9.1
$ node server.js 
{ logEnabled: true,
  port: 8888,
  responseDelay: 10,
  coils: <Buffer 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f8 2b 80 03 01 00 00 00 10 00 00 00 00 00 00 00 78 2c 80 03 01 00 00 00 04 00 00 00 00 00 00 00 00 00 ... >,
  holding: <Buffer 04 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 f8 2f 80 03 01 00 00 00 10 00 00 00 06 00 00 00 78 30 80 03 01 00 00 00 04 00 00 00 00 00 00 00 90 31 ... >,
  input: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 61 72 67 75 6d 65 6e 74 73 35 80 03 01 00 00 00 58 5c 1f 01 01 00 00 00 30 a8 ... > }
/Users/david/nodejs/jsmodbus/server.js:31
            this.on('readCoilsRequest', function (start, quantity) {
                 ^

TypeError: this.on is not a function
    at null.<anonymous> (/Users/david/nodejs/jsmodbus/server.js:31:18)
    at null.<anonymous> (/Users/david/nodejs/jsmodbus/server.js:56:9)
    at /Users/david/nodejs/jsmodbus/node_modules/stampit/dist/stampit.full.js:172:39
    at Array.reduce (native)
    at Stamp (/Users/david/nodejs/jsmodbus/node_modules/stampit/dist/stampit.full.js:171:55)
    at Object.<anonymous> (/Users/david/nodejs/jsmodbus/server.js:60:1)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
```

I did add this: `console.log(util.inspect(this));` to output the object `this`.  As you can see it doesn't have a field named `on`.

I don't grok _stampit_ but it appears to support instantiating objects in some way that other systems don't support.  Whatever.  It doesn't seem to be instantiating the object correctly in this case.

BauchBeinePoe, 2016-12-05:
Can you show me your code?

robogeek, 2016-12-05:
As I said, the code is the same as in your README.  

```
'use strict';

const stampit = require('stampit');
const modbus  = require('jsmodbus');
const util    = require('util');

var customServer = stampit()
    .refs({
        'logEnabled'        : true,
        'port'              : 8888,
        'responseDelay'     : 10, // so we do not fry anything when someone is polling this server

        // specify coils, holding and input register here as buffer or leave it for them to be new Buffer(1024)
        coils               : new Buffer(1024),
        holding             : new Buffer(1024),
        input               : new Buffer(1024)
    })
    .compose(modbus.server.tcp.complete)
    .init(function () {

        var init = function () {

            // get the coils with this.getCoils() [ Buffer(1024) ]
            // get the holding register with this.getHolding() [ Buffer(1024) ]
            // get the input register with this.getInput() [ Buffer(1024) ]

            // listen to requests

            console.log(util.inspect(this));

            this.on('readCoilsRequest', function (start, quantity) {
              console.log('readCoilsRequest', start, quantity)

                // do something, this will be executed in sync before the
                // read coils request is executed

            });

            this.on('readHoldingRegistersRequest', function (start, quantity) {
              console.log('readHoldingRegisters', start, quantity)
            });

            this.on('writeSingleCoilRequest', function (adr, value) {
              console.log('writeSingleCoil', adr, value)
            });

            // the write request have pre and post listener
            this.on('[pre][post]WriteSingleCoilRequest', function (address, value) {

            });

        }.bind(this);

        init();

    });

customServer();

// you can of course always use a standard server like so

var server = modbus.server.tcp.complete({ port : 8888 });

// and interact with the register via the getCoils(), getHolding() and getInput() calls

server.getHolding().writeUInt16BE(123, 1);
```

BauchBeinePoe, 2016-12-05:
My output:

```
{ logEnabled: true,
  port: 8888,
  responseDelay: 10,
  coils: <Buffer 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b8 35 e0 03 00 00 00 00 10 00 00 00 00 00 00 00 38 36 e0 03 00 00 00 00 04 00 00 00 00 00 00 00 00 00 ... >,
  holding: <Buffer 04 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 b8 39 e0 03 00 00 00 00 10 00 00 00 06 00 00 00 38 3a e0 03 00 00 00 00 04 00 00 00 00 00 00 00 50 3b ... >,
  input: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 61 72 67 75 6d 65 6e 74 73 36 e0 03 00 00 00 00 d8 19 4c 01 00 00 00 00 60 d1 ... >,
  emit: [Function],
  on: [Function],
  once: [Function],
  removeListener: [Function],
  log: 
   { error: [Function: bound ],
     warn: [Function: bound ],
     info: [Function: bound ],
     debug: [Function: bound ],
     trace: [Function: bound ] },
  logInfo: [Function: bound ],
  logError: [Function: bound ],
  logLevel: 'info',
  onData: [Function: bound ],
  setRequestHandler: [Function],
  getCoils: [Function],
  getInput: [Function],
  getHolding: [Function],
  inState: [Function],
  getState: [Function],
  setState: [Function],
  close: [Function],
  hostname: '0.0.0.0' }

```

What version are you using? Have you tried the SimpleServer.js example in the example/tcp folder?

robogeek, 2016-12-05:
Hurm... 

As I said above,

```
$ node --version
v6.9.1
```

The SimpleServer.js is not much different from this ... I didn't see any significant difference and therefore did not try it.

BauchBeinePoe, 2016-12-06:
I am sorry, I ment what jsmodbus version are you using?

robogeek, 2016-12-06:
```
  "dependencies": {
    "jsmodbus": "^2.0.0",
    "stampit": "^3.0.6"
  }
```

BauchBeinePoe, 2016-12-06:
You are using the wrong stampit module.

From the package.json:

```
"stampit": "^2.1.2"
```

robogeek, 2016-12-06:
Okay -- I removed the stampit dependency from my package.json and now it's working correctly.

BTW your client example in the README has two syntax errors:

Line 75: `client.writeSingleRegister(13, Buffer.from([0x00, 0x2A])).then(function (resp) {`

There was a missing comma

Line 103: `client.writeMultipleRegisters(4, Buffer.from([0x00, 0x01, 0x00, 0x02, 0x00, 0x03, 0x00, 0x04])).then(function (resp) {`

There was a missing `)`

BauchBeinePoe, 2016-12-06:
Glad to hear.

The readme has basically a documentary purpose. It's usually not intended for copy and paste. That's what I made the examples for. 

But I will change the typos on the next update. Thanks :-)

closed 2016-12-06 by BauchBeinePoe