Cloud-Automation/node-modbus #106: reqId out of range error in modbus-tcp-client.js

jbockerstette, 2016-12-09:
In the onSend method of modbus-tcp-client.js, you have the following code. If you let this code run long enough, you will get a TypeError: "value" argument is out of bounds. The problem is that the var reqId += 1 will grow forever. Once this value gets beyond 65535, the method head.writeUInt16BE(reqId, 0) does a checkInt on the reqId and throws TypeError. At some point, you need to set this back to zero.

```
var onSend = function (pdu) {
      this.log.debug('Sending pdu to the socket.')

      reqId += 1

      var head = Buffer.allocUnsafe(7)

      head.writeUInt16BE(reqId, 0)
 
```

Here is the stack trace:
```

TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1026:11)
    at Buffer.writeUInt16BE (buffer.js:1095:5)
    at Object.<anonymous> (/home/rppuser/dev/workspace/webstorm/rpp-modbus-client/node_modules/jsmodbus/src/modbus-tcp-client.js:129:12)
    at Object.emit (/home/rppuser/dev/workspace/webstorm/rpp-modbus-client/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:38)
    at Object.<anonymous> (/home/rppuser/dev/workspace/webstorm/rpp-modbus-client/node_modules/jsmodbus/src/modbus-client-core.js:60:12)
    at Object.emit (/home/rppuser/dev/workspace/webstorm/rpp-modbus-client/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:38)
    at Object.setState (/home/rppuser/dev/workspace/webstorm/rpp-modbus-client/node_modules/stampit-state-machine/src/stampit-state-machine.js:23:14)
    at Object.<anonymous> (/home/rppuser/dev/workspace/webstorm/rpp-modbus-client/node_modules/jsmodbus/src/modbus-tcp-client.js:54:12)
    at emitNone (events.js:91:20)
    at Socket.emit (events.js:185:7)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1072:10)

```

BauchBeinePoe, 2016-12-09:
I am pretty sure I added this with a modulo at some time. Must have been overwritten. Thanks @jbockerstette I'll make a pull request and fix this.

psorowka, 2016-12-09:
very good catch! Shit, once again: my fault. When changing from `Put` to `Buffer` i should have introduced a modulo. The `Put` was doing that implicitly due to its implementation. @BauchBeinePoe are you on it? This is serious cause it will make any client crash after 65535 msgs... :( I'd do it, too

BauchBeinePoe, 2016-12-09:
Shit happens @psorowka. If you want to do it, be my guest. Can we provide a test for this?

psorowka, 2016-12-09:
we must :)

psorowka, 2016-12-09:
@BauchBeinePoe while I'm doing that I have a suggestion: after having added TravisCI it would also be worth to add a Coverage tool. Coveralls.io is free for OS projects and does a very good job on it. By examing the coverage locally with istanbul I found that especially the `modbus-tcp-client.js` has extremely poor coverage meaning that a lot of code in there (and the passage in question is included) are not executed in a single test.

The Coverage report is a nice indicator to come by that. What do you think?

psorowka, 2016-12-09:
see #107

psorowka, 2016-12-10:
@BauchBeinePoe here is a side-note. In the 1.2.x. branch we still have the old behavior:

```
            reqId += 1;

            var pkt = Put()
                .word16be(reqId)                 // transaction id
                .word16be(this.protocolVersion) // protocol version
                .word16be(pdu.length + 1)        // pdu length
                .word8(this.unitId)	             // unit id
                .put(pdu)                        // the actual pdu
                .buffer();

            currentRequestId = reqId;

            socket.write(pkt);
```

As said, the `word16be(reqId)` performs the modulo implicitly due to its implementation and thus does not lead to the above error. However, that means that the pdu will always contain a reqId < 2^16 while the value of the variable reqId is increasing. That means that after the first 2^16 messages the value of reqId and the actually sent requestId in the pdu start to differ and consequently the value of `currentRequestId` is not the same as the sent requestId. That means the whole trash logic (which I understand at first glance to be honest) will only work for the first 2^16 messages. The current behavior in my PR #107 would solve that. I don't know how critical that is and if a backport of the bugfix to the 1.2.x is required.

BauchBeinePoe, 2016-12-11:
I think the issue on the 1.2.x branch is critical One so i will make a pull request for this.

BauchBeinePoe, 2016-12-11:
I will dig into coveralls.io when I am in the office. I am not sure how it checks what needs to be covered.

psorowka, 2016-12-11:
@BauchBeinePoe it basically works by observing which code lines are actually executed while running the tests. Coveralls specifically hooks into travis for that.

closed 2016-12-11 by psorowka

BauchBeinePoe, 2016-12-12:
Fixed this also in 1.2.7.