Cloud-Automation/node-modbus #111: It didn't connect the server.I need help!

panyunyi, 2017-01-12:
Hello,sorry to bother you!
I really need some help.
[image]

modscan32 is worked in this config.But jsmodbus didn't work.

panyunyi, 2017-01-12:
[image]

BauchBeinePoe, 2017-01-16:
Are you sure you connect to the right port and ip address? The timeout can be from the tcp connection since you have set auto reconnect to true.

panyunyi, 2017-01-17:
Yes,I'm sure connect to the right port and ip address.The second picture is modscan.They have same config.
[image]
[image]

BauchBeinePoe, 2017-01-17:
Can you post the code here, so that I can reproduce.

panyunyi, 2017-01-17:
var modbus = require('jsmodbus');

// create a modbus client
var client = modbus.client.tcp.complete({ 
        'host'              : '192.168.1.8', 
        'port'              : 10000,
        'autoReconnect'     : false,
        'reconnectTimeout'  : 1000,
        'timeout'           : 5000,
        'unitId'            : 1
    });

client.connect();

// reconnect with client.reconnect()

client.on('connect', function () {

    client.readInputRegisters(2, 10).then(function (resp) {

        // resp will look like { fc: 4, byteCount: 20, register: [ values 0 - 10 ], payload: <Buffer> }
        console.log(resp);

    }, console.error);

});

client.on('error', function (err) {

    console.log(err);

})

BauchBeinePoe, 2017-01-17:
The best way to solve these kind of problems is to do it step by step. Try the code below. There are no modbus requests, just the connection. I would bet that this fails. If so, check your server and firewall settings.

````
var modbus = require('jsmodbus');

// create a modbus client
var client = modbus.client.tcp.complete({
    'host' : '192.168.1.8',
    'port' : 10000,
    'autoReconnect' : false,
    'reconnectTimeout' : 1000,
    'timeout' : 5000,
    'unitId' : 1
});

client.connect();

client.on('connect', function () {
    console.log("Connected!");
});

client.on('error', function (err) {
    console.log("Connection error!", err);
})
````

panyunyi, 2017-01-17:
OK!Thank you for your patience.I will try.

panyunyi, 2017-01-18:
Hi...It's connected.But the readInputRegisters didn't work.I think it maybe the response code is wrong.
[image]
[image]
The right code is:
0000 0000 0006 0104 0001  0014
How should I modify the demo code to success?

panyunyi, 2017-01-18:
[image]
It's worked!Thank you!

closed 2017-01-18 by BauchBeinePoe