Cloud-Automation/node-modbus #114: What's wrong with the writeMutipleRegisters?

panyunyi, 2017-01-30:
I used the writeSingleRegister and it's worked.
`client.writeSingleRegister(1, 42).then(function (resp) {

        // resp will look like { fc: 6, byteCount: 4, registerAddress: 13, registerValue: 42 }
        console.log(resp);

    }, console.error);`
[image]
It's not what I want when I used the writeMutipleRegisters.
`client.writeMutipleRegisters(1, [1,2,3,4]).then(function (resp) {

        // resp will look like { fc: 6, byteCount: 4, registerAddress: 13, registerValue: 42 }
        console.log(resp);

    }, console.error);`
[image]

BauchBeinePoe, 2017-01-30:
Why is it a float that is presented there? Usually the Modbus register are 16bit unsigned integer.

panyunyi, 2017-01-30:
Yes,there are float numbers.
I get environmental information through Modbus.The environmental information collector provides float numbers.
[image]
[image]
[image]
[image]
[image]
Is there any way to write float numbers?

BauchBeinePoe, 2017-01-30:
The Readme says 

````
client.writeMultipleRegisters(4, [1, 2, 3, 4]).then(...)
````

it should actually look like this 

````
client.writeMultipleRegisters(4, Buffer.from([1, 2, 3, 4])).then(...)
````

can you try this?

psorowka, 2017-01-30:
I would second @BauchBeinePoe on this. When passing an array with numbers to the function, they will be interpreted as Big Endian unsignedInt16 and will not result in the float values you expect.

However, @BauchBeinePoe I think your second example is not correct:

1. the Buffer must contain 8 Bytes (4 Words), thus should be [0,1,0,2,0,3,0,4]
1b. float32 is even 4 byte long so could potentially even require 16 Bytes!
2. still the number representation is not float. Better go this way:

```
let buf = Buffer.allocUnsafe(16)
buf.writeFloatBE(1,0)
buf.writeFloatBE(2,4) // <-- skip 4 byte!
buf.writeFloatBE(3,8)
buf.writeFloatBE(4,12)
```

For number representation look at this: 

```
b = Buffer.allocUnsafe(4)
b.writeUInt16BE(1,0)
--> <Buffer 00 01 00 00>
b.writeFloatBE(1,0)
--> <Buffer 3f 80 00 00>
```

still, I am a bit wondering why the `registerAddressRaw` and `registerValueRaw` fields are not looking good in the first sample... have to look at this later.

BauchBeinePoe, 2017-01-30:
Right, I was confused with the changes in the API from Array to Array/Buffer.

panyunyi, 2017-01-30:
It's succeed!
Thank you for  your help! @psorowka @BauchBeinePoe

BauchBeinePoe, 2017-01-30:
@panyunyi Would be great if you can say what went wrong so others can relate to this issue.

panyunyi, 2017-01-30:
In my case,I would like to write float numbers.But the result of writeMultipleRegisters is not what I want.
The right way is what @psorowka said.
Define a Buffer and then writeFloatBE.
```
let buf=Buffer.allocUnsafe(16);
buf.writeFloatBE(1,0);
buf.writeFloatBE(2,4); // <-- skip 4 byte!
buf.writeFloatBE(3,8);
buf.writeFloatBE(4,12);
let bufs=[];
for(let i=0;i<buf.length/4;i++){
	let arr1=buf.slice(i*4,i*4+4);
	let temp1=arr1[0];
	let temp2=arr1[1];
	arr1[0]=arr1[2];
	arr1[1]=arr1[3];
	arr1[2]=temp1;
	arr1[3]=temp2;
	bufs.push(arr1);
}			
let buf2=Buffer.concat(bufs);
client.writeMultipleRegisters(1, buf2).then(function (resp) {
	console.log(buf2);
	console.log(resp);
  }, console.error).finally(function () {
	client.close();
});
```
[image]
I recombined the buf array.Because my device needs a special endianess.
Like this:
Usually:
```
b.writeFloatBE(1,0)
--> <Buffer 3f 80 00 00>
```
Edianess of my device(It is not LE):
```
--> <Buffer  00 00 3f 80>
```

closed 2017-01-30 by BauchBeinePoe

marcosps, 2017-06-10:
This sure helped me, thanks a lot @stefanpoeter and @psorowka and @panyunyi for asking this here!

awadekaustubh, 2017-08-26:
Can anyone explain how to read float values from holding register? In file ReadHoldingRegisters.js , I am unable to understand this piece of code    
var fc = 3
var pdu = Buffer.allocUnsafe(3)
pdu.writeUInt8(fc)
pdu.writeUInt16BE(start, 1)
pdu.writeUInt16BE(quantity, 3)

What changes needs to be done so that I can read float values?

blhough, 2017-08-26:
@awadekaustubh 
Try something like this
```
client.readHoldingRegisters(addr, num).then((resp)=> {
  console.log(new Float32Array(resp.swap32().buffer));
});
```
where `num` is a multiple of 2 

`swap32()` is necessary because modbus is typically big-endian while js is little-endian.

In the code you posted it looks like that is just setting up the modbus frame which is part of the protocol. No changes need to be made to the source code to be able to handle floats.

Modbus uses 16bit registers while a float (IEEE 754) is 32bits. So to write and read floats 2 registers are needed.

kotopuz, 2019-01-20:
Dear all,

Somehow that was not working for me.

I've found answer in stack overflow and modified that. TESTED!
https://stackoverflow.com/questions/6887003/convert-32bit-unsigned-real-data-type-splitted-into-two-16-bit-signed-words/54272568#54272568

```
//buffer to real conversion and back BIG ENDIAN, FLOAT is NOT SWAPPED, FLOAT 32 BIT require 2 16 bit array elements
function uint16ToFloat32(uint16array) {
    var buffer = new ArrayBuffer(4);
    var intView = new Uint16Array(buffer);
    var floatView = new Float32Array(buffer);
  
    intView[0] = uint16array[0];
    intView[1] = uint16array[1];
    return floatView[0];
  }
//use it: console.log("Converted ints to", uint16ToFloat32(18584, 18081));
```

```
function float32ToUint16(float32) {
    var buffer = new ArrayBuffer(4);
    var intView = new Uint16Array(buffer);
    var floatView = new Float32Array(buffer);
  
    floatView[0] = value;
    return [intView[0], intView[1]];
  }
//use it: console.log("Converted float to", float32ToUint16(20644.297));
```