Cloud-Automation/node-modbus #115: port remains in LISTEN

roccomuso, 2017-01-30:
If I run the example script on a given port, and then I close it, I still have:

    tcp        0      0 0.0.0.0:8889            0.0.0.0:*               LISTEN      30432/tc        

And I cannot run again the script because of the `EADDRINUSE` error.

BauchBeinePoe, 2017-01-30:
I cannot confirm this. When I start the SimpleServer.js script `lsof -i :8888` gives me the following

````
COMMAND  PID   USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
node    8720 stefan   12u  IPv4 184113      0t0  TCP *:8888 (LISTEN)
````

When I quit the script, no process is listening on this port and I can restart the server without errors.

The example script listens to port 8888, did you change the port because it says 8889?

roccomuso, 2017-01-30:
I changed the port because the `8888` was still in listen :') Now the `8888` is no more listening but there's the `8889` busy.

BauchBeinePoe, 2017-01-30:
Can you do a `ps -e` and confirm that the process is really killed? Are you polling the server with a modbus client?

roccomuso, 2017-01-30:
The process is running. If I kill the process, it's respawned with another PID.

BauchBeinePoe, 2017-01-30:
I do not see anything in the SimpleServer.js file that restarts the script. I'll do the following to start the script:

1. Clone the node-modbus repo.
2. cd into the node-modbus folder.
3. node examples/tcp/SimpleServer.js
4. lsof -i :8888 (from another terminal gives me the port and the script)
5. Wait a few seconds maybe make a request from a modbus client.
6. CTRL+C (SigInt) - Server quits.
7. lsof -i :8888 (gives me nothing)
8. Done.

Did you alter anything else in the script?

roccomuso, 2017-01-30:
Nevermind.. My fault. It was a client issue, respawning the `SimpleServer.js`.
Thanks for the fast support :)

closed 2017-01-30 by roccomuso