Cloud-Automation/node-modbus #117: Feature request: make rtu / serial support optional

flottokarotto, 2017-02-02:
serialport is the only native dependency.
jsmodbus would be a js-only modbus solution without it.

BauchBeinePoe, 2017-02-02:
There is an option for npm packages to declare certain dependencies as optional. This way the module will be installed even thou this optional module is missing or it's installation routine failed. So this could be done. Why do you want it to be js-only?

flottokarotto, 2017-02-03:
For cross-build environments (e.g. open embedded), it is necessary to delete all native dependencies inside node_modules (or complete node_modules). A js-only solution would be a lot easier to deploy.

BauchBeinePoe, 2017-02-16:
You can always fork this project without the native dependencies. Since serial is already introduced here I won't take it back, sorry.

closed 2017-02-16 by BauchBeinePoe

GermanBluefox, 2017-02-16:
I must say, that would to prefer to move serialport to optional
dependencies.
It is really optional an as far as I know almost no one use this node as
serial rtu. I want to prepare pull request to fix chanking with serial.

If you move serialport to optional you loose nothing.  The people who wants
to use this packet with serial port, they will care about properly
installation. And the people who wants to use just over tcp they are just
happy. My platform runs on very different devices, nas, arm5, and so on.
The world is not only raspi, Mac and PC.

stefanpoeter, 2017-05-03:
See branch v3.0.0-alpha:
The serialport is no longer a dependency. Yet you are able to create a rtu (serial) client.