Cloud-Automation/node-modbus #118: How could I broadcast data of modbus in realtime?

panyunyi, 2017-02-06:
I used websockets(socket.io) to transfer the data.But the app crashed after a period of time after because of memory overflow caused the collapse.
```
var io=require('socket.io')(http);
var modbus = require('jsmodbus');
io.sockets.on( "connection", function( socket ){
	console.log('socket connect');
	let client = modbus.client.tcp.complete({ 
        'host'              : '192.168.1.8', 
        'port'              : 10000,
        'autoReconnect'     : false,
        'reconnectTimeout'  : 1000,
        'timeout'           : 5000,
        'unitId'            : 1
	});
	client.connect();
	client.on('connect', function () {
		console.log('modbus');
		setInterval(function(){
			client.readInputRegisters(1,24).then(function(resp){
					let arr=resp.payload;
					socket.emit('info',arr);
				},console.error);
		},1000);
		socket.on('switch',function(msg){
			console.log(msg.flag+" "+msg.on);
			client.writeSingleCoil(msg.flag, msg.on).then(function (resp) {
				console.log(resp);
			}, console.error);
			console.log(msg);
		});
		socket.on('disconnect',function(socket){
			console.log('socket disconnect');
			client.close();
			delete client;
		});
	});
	client.on('error', function (err) {
		console.log(err);
	});
	client.on('close', function () {
		console.log('modbus close');
	});
});
```
[image]

psorowka, 2017-02-06:
Chances are good that the memory exceed comes from an outgrowing modbus command queue. This can happen in two ways:

1. you are polling faster than your client responds
2. you continue polling when the client is disconnected. jsmodbus will queue all requests you issue when the client is disconnected and send them out once it is connected again.

In your code, I see that you don't handle the case 2 correctly so I would search there first.

Still, from my experience it is not recommendable to poll modbus with a `setInterval` but rather start the next read in the callback of the previous.

BauchBeinePoe, 2017-02-06:
@panyunyi See issue #19 and #29 for some inspiration.

panyunyi, 2017-02-12:
I changed my code.Added  a line.Look at  this:
```
socket.on('disconnect',function(socket){
			console.log('socket disconnect');
			client.close();
			**clearInterval(interval);**
			delete client;
		});
```
The anonymous timer function registered by setInterval holds a reference to the client. In the case of an anonymous function can not be recovered (setInterval is a global function), client can not be recycled.And then memory leaks.
So I added the clearInterval to clear the global function when the socket is disconnected.
It's worked.
There have been no memory leaks in the past.
But there are still some print error messages:
Is this mean that polling faster than the client's response?

psorowka, 2017-02-12:
no this looks like a race condition between client.close and clearInterval. I would first switch the two function calls and second I think this could be a good case for `process.nextTick`. The `delete client` should IMHO not be necessary.

```
socket.on('disconnect',function(socket){
			console.log('socket disconnect');
			clearInterval(interval);
                        process.nextTick(client.close);
		});
```

panyunyi, 2017-02-12:
@psorowka I will try it.Thank you very much!

BauchBeinePoe, 2017-02-16:
@panyunyi Any progress?

panyunyi, 2017-02-17:
@BauchBeinePoe Yes.It's looks like ok.But the error is still here.

```
socket.on('disconnect',function(socket){
			console.log('socket disconnect');
			//client.close();
			clearInterval(interval);
			process.nextTick(client.close);
		});
```
```
Error:write after end
   at writeAfterEnd(_stream_writeable.js:193:12)
   at Socket.Writeable.write(_stream_writeable.js:244:5)
   at Socket.write(net.js:658:40)
```
The good news is the error has been appeared less.

BauchBeinePoe, 2017-02-17:
Try this.

````
client.on('connect', function () {

                var disconnected = false;

		console.log('modbus');
		setInterval(function(){
			client.readInputRegisters(1,24).then(function(resp){
					let arr=resp.payload;
                                        if (disconnected) {
                                                return;
                                        }
					socket.emit('info',arr);
				},console.error);
		},1000);

...

socket.on('disconnect', function (socket) {
                              console.log('socket disconnect');
                              disconnected = true;
                              ... });

````

panyunyi, 2017-02-17:
OK.I will try it.We will see the results the day after tomorrow.
Thank you very much!

panyunyi, 2017-02-17:
[image]
There is something wrong.
```
client.on('connect', function () {
		var disconnected=false;
		console.log('modbus');
		let interval=setInterval(function(){
			client.readInputRegisters(1,24).then(function(resp){
					let arr=resp.payload;
					if(disconnected){
						return;
					}
					socket.emit('info',arr);
				},console.error);

```
```
socket.on('disconnect',function(socket){
			console.log('socket disconnect');
			disconnected=true;
			clearInterval(interval);
			process.nextTick(client.close);
		});
```

BauchBeinePoe, 2017-02-24:
Most of the time this is due to a missing execution context for the called method. Seems like you need to apply the `client` object to the `process.nextTick(client.close)`. Maybe like this `process.nextTick(client.close.bind(client))`.

panyunyi, 2017-02-24:
It's happened again.The error is still here.
But it doesn't stop ever.

panyunyi, 2017-02-24:
Thank you very much for your help!

closed 2017-02-28 by BauchBeinePoe