Cloud-Automation/node-modbus #120: input registers

GermanBluefox, 2017-02-11:
I am completely confused about naming.
Don't you want to create 2.x because of input registers (not discrete)?

What the name should I use to include input registers?

psorowka, 2017-02-12:
I don't get it.

GermanBluefox, 2017-02-12:
We wanted to extend the Server Register types from:
- Discrete inputs
- Holding Registers
- Coils

to support "Input registers" non binary read only 16 bits long.

But I cant find any changes concerning this.

As I can remember, the change to 2.x from 1.x because of breaking changes in naming and we wanted to add "Input registers". But because the "discrete inputs" had a name just "inputs", it was the problem to add real "input registers". 

And someone wanted to rename "inputs" (discrete) to "discrete".

And my question how "input registers" should be named now???

psorowka, 2017-02-12:
So you are referring to FC4? This is supported by client and server. No need to make any names up, they are quite well defined in the modbus spec, arent they? 

I think you mix something up, the change to 2.x was made after a big PR of mine in which I (beneath other stuff) exchanged the promise library from Q to Bluebird which led to some interface changes. additionally, the Buffer type argument handling was introduced in this step.

I still don't really get which function you want to rename (or introduce)

GermanBluefox, 2017-02-12:
There are 4 types of registers, but server supports only 3. I want to add
forth - input register (not discrete). Or am I wrong and just can't find it?

psorowka, 2017-02-12:
According to the source, both `ReadInputRegisters` (fc4) (https://github.com/Cloud-Automation/node-modbus/blob/master/src/handler/server/ReadInputRegisters.js#L36) and `ReadDiscreteInputs` (fc2) (https://github.com/Cloud-Automation/node-modbus/blob/master/src/handler/server/ReadDiscreteInputs.js#L38) refer to the same Memory Buffer.

If I understand correct, this is the issue you address? I totally agree, it seems strange. Is there reason behind it @BauchBeinePoe ?

GermanBluefox, 2017-02-13:
> master/src/handler/server/ReadInputRegisters.js#L36) and
> Automation/node-modbus/blob/master/src/handler/server/
> ReadDiscreteInputs.js#L38) refer to the same Memory Buffer.

Yes. You are right.

BauchBeinePoe, 2017-02-13:
If I remember correctly, the Modbus specification does not define the kind of memory areas the function Codes address. That seemed to me as being a developer decision. I always looked at the discrete Input register function as a bitwise access for the input registers. 

But I agree, it does not have to be this way. There can also be a fourth buffer for the discrete inputs. @GermanBlueFox, wasn't there a pull request that already addresses this issue?

GermanBluefox, 2017-02-13:
@BauchBeinePoe Yes it was a pull request. But I rewrote it. 
https://github.com/GermanBluefox/node-modbus/blob/discrete-inputs-germanbluefox/src/modbus-server-core.js
I called them:
- "inputs" - input registers
- "discrete" - discrete inputs

If it is ok I will rename it in tests and examples too. But.... It is break change.

BauchBeinePoe, 2017-02-16:
Create a pull request and then we will see. :-)

BauchBeinePoe, 2017-02-21:
I'll close this since it is not really an issue since you want to do a pull request and I don't like issues just hanging around for no reason :-)

closed 2017-02-21 by BauchBeinePoe