Cloud-Automation/node-modbus #121: Connection is closed [proposed Label] Question

bladerunner2020, 2017-02-21:
I have established connection to Ethernet I/O Module from ICPDAS and I noticed that the connection is regularly closed by itself. I added client.reconnect and it seems working ok. 

`client.on('close', function () {
    client.reconnect();
});`

I have few questions.

1. On receiving http-request I call writeSingleCoil. Currently it's theoretically possible that writeSingleCoil could be called when client is not connected. I could add flag isConnected and check it, but maybe there is a better way to assure that client is connected?

2. Is it a good practice to keep connection alive or is it preferred to open/close connection on each requests?

BauchBeinePoe, 2017-02-21:
First of all I would inspect why the connection fails. Is it the server or the client that closes the connection.

1. Currently there is no flag in the client whether the client is connected or not.
2. That depends on your application. How many requests do you expect will be made to the modbus server? if you only do a write request from time to time then open a new connection, write and close it again. If you are monitoring modbus data and want to distribute it to multiple clients then do this via a single modbus connection.

bladerunner2020, 2017-02-21:
Thanks. I have not many requests so I think open/write/close approach could be used in my case. There are couple of cons - I would like to reduce the reaction time and write to modbus as soon as possible (how quick is client.connect?) and the second reason (most important), that there could be 2 near simultaneous requests (in this case open-write-close-open-write-close - is not the best way to handle them).

Any ideas how could I find out why the connection fails?

BauchBeinePoe, 2017-02-24:
You can enable logging in the client. I do not have any data on the speed of connection, but it is a regular tcp/ip socket that is set up. If you want to have some data you can always use `console.time` and `console.timeEnd` to measure something (it works in the browser at least, not sure if node.js supports it too)

bladerunner2020, 2017-02-27:
I checked: the connection is closed every 89895 ms or so...

bladerunner2020, 2017-02-27:
Console output with NET module debug on.

```
NET 51075: onread -4095
NET 51075: EOF
NET 51075: onSocketEnd ReadableState {
  objectMode: false,
  highWaterMark: 16384,
  buffer: BufferList { head: null, tail: null, length: 0 },
  length: 0,
  pipes: null,
  pipesCount: 0,
  flowing: true,
  ended: true,
  endEmitted: false,
  reading: false,
  sync: false,
  needReadable: false,
  emittedReadable: false,
  readableListening: false,
  resumeScheduled: false,
  defaultEncoding: 'utf8',
  ranOut: false,
  awaitDrain: 0,
  readingMore: false,
  decoder: null,
  encoding: null }
NET 51075: onSocketFinish
NET 51075: oSF: ended, destroy ReadableState {
  objectMode: false,
  highWaterMark: 16384,
  buffer: BufferList { head: null, tail: null, length: 0 },
  length: 0,
  pipes: null,
  pipesCount: 0,
  flowing: true,
  ended: true,
  endEmitted: false,
  reading: false,
  sync: false,
  needReadable: false,
  emittedReadable: false,
  readableListening: false,
  resumeScheduled: false,
  defaultEncoding: 'utf8',
  ranOut: false,
  awaitDrain: 0,
  readingMore: false,
  decoder: null,
  encoding: null }
NET 51075: destroy undefined
NET 51075: destroy
NET 51075: close
NET 51075: close handle
NET 51075: destroy undefined
NET 51075: destroy
NET 51075: already destroyed, fire error callbacks
NET 51075: emit close
== Modbus: connection closed.
== Modbus connected.
NET 51075: pipe false undefined
NET 51075: afterConnect
NET 51075: _read
NET 51075: Socket._read readStart
```

bladerunner2020, 2017-03-14:
I suppose that modbus device closes connection if there is not activity. I am planning to try reading call  every minute and see if this issue is gone or not.

stefanpoeter, 2017-03-21:
Since this is related to #122 I'll close this one.

closed 2017-03-21 by stefanpoeter