Cloud-Automation/node-modbus #122: When .fail is called? [proposed Label] Question

bladerunner2020, 2017-02-21:
I have another question.

Time to time I am getting ECONNRESET error on writeSingleCoil. 
This error I get only when run my project on OSX, haven't encountered it on Windows.

```
    client.writeSingleCoil(id, value)
        .then()
        .fail(function (err) {
            console.log(err);
        })
        .done(function (resp, err) {
            console.log("success. ", resp);
        });
```

I found that .fail is not called in this case. So the question is - when .fail is called? 

PS
If anybody have ideas how to solve ECONNRESET I would appreciate as well.

BauchBeinePoe, 2017-02-21:
Can you show me the stack trace from the error message?

bladerunner2020, 2017-02-22:
Here it is. If you would like I could turn on debug output for net module.
```
{ Error: read ECONNRESET
    at exports._errnoException (util.js:1022:11)
    at TCP.onread (net.js:569:26) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
```

BauchBeinePoe, 2017-02-24:
I am not sure if the client actually has enough information to call the `.fail` method. Seems like there is an ECONNRESET when the net module is trying to read data from the socket. Are you listening to the `close` event?

More information would be useful. Can you enable logging in the client and the net module?

bladerunner2020, 2017-02-27:
Here it's. Surprisingly, this time I got ECONNRESET after 2 successful attempts of writeSingleCoil. Typically there should be no NET activity after  { fc: 5, outputAddress: 5, outputValue: false }...

```
== writing coil reg = 5  value =1
NET 47587: afterWrite 0
NET 47587: afterWrite call cb
== { fc: 5, outputAddress: 5, outputValue: true }
NET 47587: onread 12
NET 47587: got data
NET 47587: _read
== writing coil reg = 5  value =0
NET 47587: onread 12
== { fc: 5, outputAddress: 5, outputValue: false }
NET 47587: got data
NET 47587: _read
NET 47587: _onTimeout
NET 47587: destroy undefined
NET 47587: destroy
NET 47587: close
NET 47587: close handle
NET 47587: emit close
{ Error: socket hang up
    at createHangUpError (_http_client.js:254:15)
    at Socket.socketCloseListener (_http_client.js:286:23)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at TCP._handle.close [as _onclose] (net.js:498:12) code: 'ECONNRESET' }
```

bladerunner2020, 2017-02-27:
Yes, I am listening to the close event. It's not called in this case.

bladerunner2020, 2017-02-27:
One more. In this case ECONNRESET and disconnection are in the same time interval. The close event is called but it's always called when connection is closed and restored without error. In this case onread, got data, _read came after { fc: 5, outputAddress: 5, outputValue: false }...

```
== writing coil reg = 5  value =1
NET 48291: afterWrite 0
NET 48291: afterWrite call cb
NET 48291: onread -54
NET 48291: destroy
NET 48291: close
NET 48291: close handle
NET 48291: destroy undefined
NET 48291: destroy
NET 48291: already destroyed, fire error callbacks
NET 48291: emit close
Mobdus error.  { Error: read ECONNRESET
    at exports._errnoException (util.js:1022:11)
    at TCP.onread (net.js:569:26) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
== Modbus: connection closed.
== Modbus connected.
NET 48291: pipe false undefined
NET 48291: afterConnect
NET 48291: _read
NET 48291: Socket._read readStart
== writing coil reg = 5  value =0
== { fc: 5, outputAddress: 5, outputValue: false }
NET 48291: onread 12
NET 48291: got data
NET 48291: _read
```

BauchBeinePoe, 2017-03-03:
What jsmodbus are you using?

bladerunner2020, 2017-03-04:
"jsmodbus": "^1.2.7"

BauchBeinePoe, 2017-03-04:
At first I thought this was #107 but this should be fixed in v1.2.7. I am not sure which side is closing the connection. Is there a way to determine that? Are you polling data like constantly? Or do you just keep the tcp connection open?

BauchBeinePoe, 2017-03-13:
Can you check if the error occures with v2.0.2?

stefanpoeter, 2017-05-02:
Or with v3.0.0-alpha (new branch)

closed 2017-05-02 by stefanpoeter