Cloud-Automation/node-modbus #124: npm package is out of date

brandly, 2017-03-10:
latest [on npm](https://www.npmjs.com/package/jsmodbus): `v1.2.7`

latest [in this repo](https://github.com/Cloud-Automation/node-modbus/blob/master/package.json#L3): `v2.0.2`

BauchBeinePoe, 2017-03-13:
By version, yes! But I maintain the v1.2.x branch for critical errors. v1.2.7 was the last published version after v2.0.2. Don't know why, but npm shows and offers the latest commited version and not the highest version number. I'll look into it. But you can download the latest version by typing `npm install jsmodbus@2.0`.

brandly, 2017-03-13:
> npm shows and offers the latest commited version and not the highest version number

i had no idea!

thanks @BauchBeinePoe!

closed 2017-03-13 by brandly