Cloud-Automation/node-modbus #125: Error using this module on webpack

Jotan, 2017-03-13:
Hello, a I'm having problem using this module on webpack any suggestion? , here is the error :

`fs.js:951Uncaught Error: ENOENT: no such file or directory, scandir 'D:\handler\client'fs.readdirSync @ fs.js:951fs.readdirSync @ ELECTRON_ASAR.js:576(anonymous function) @ modbus.js?3944:23(anonymous function) @ modbus.js?3944:46(anonymous function) @ build.js:1327__webpack_require__ @ build.js:556fn @ build.js:87(anonymous function) @ main.js?17f6:13(anonymous function) @ build.js:1267__webpack_require__ @ build.js:556fn @ build.js:87(anonymous function) @ build.js:588__webpack_require__ @ build.js:556(anonymous function) @ build.js:579(anonymous function) @ build.js:582`

BauchBeinePoe, 2017-03-13:
This does not seem related with jsmodbus. Do you have some code for me?

Jotan, 2017-03-13:
Looking in jsmodbus folder, modbus.js on line 23 have: `
fs.readdirSync(path.join(__dirname, '/handler/client'))` I don't how can I change that for webpack

Jotan, 2017-03-14:
I solved just need to add on webpack.config.js this:
`  node: {
  __dirname: true
},`

Thanks for helping and for this great package!

closed 2017-03-14 by BauchBeinePoe