Cloud-Automation/node-modbus #129: ERROR: Client in Error State

NoobTW, 2017-03-30:
I was able to connect and read coils using jsmodbus 0.5.2, but I cannot read coils using this version. Below is the log output.

I tried in node v7.8.0 and v6.9.1; both not worked.
```
2017-03-30T08:44:57.443Z 'DEBUG :' 'init' 'connect'
2017-03-30T08:44:57.529Z 'DEBUG :' 'connect' 'connect'
2017-03-30T08:44:57.610Z 'DEBUG :' 'connect' 'ready'
2017-03-30T08:44:57.612Z 'DEBUG :' 'Trying to flush data.'
2017-03-30T08:44:57.621Z 'DEBUG :' 'ready' 'waiting'
2017-03-30T08:44:57.623Z 'DEBUG :' 'Sending pdu to the socket.'
2017-03-30T08:44:57.629Z 'DEBUG :' 'Data flushed.'
2017-03-30T08:45:02.636Z 'ERROR :' 'Request timed out.'
2017-03-30T08:45:02.638Z 'DEBUG :' 'waiting' 'error'
2017-03-30T08:45:02.640Z 'ERROR :' 'Client in error state.'
{ err: 'timeout' }
2017-03-30T08:45:02.672Z 'DEBUG :' 'Socket closed with error' false
2017-03-30T08:45:02.675Z 'DEBUG :' 'error' 'closed'
2017-03-30T08:45:02.677Z 'DEBUG :' 'Clearing timeout of the current request.'
2017-03-30T08:45:02.680Z 'DEBUG :' 'Cleaning up request fifo.'
2017-03-30T08:45:03.684Z 'DEBUG :' 'closed' 'connect'
2017-03-30T08:45:03.694Z 'DEBUG :' 'connect' 'ready'
2017-03-30T08:45:03.696Z 'DEBUG :' 'Trying to flush data.'
2017-03-30T08:45:03.701Z 'DEBUG :' 'ready' 'waiting'
2017-03-30T08:45:03.703Z 'DEBUG :' 'Sending pdu to the socket.'
2017-03-30T08:45:03.707Z 'DEBUG :' 'Data flushed.'
2017-03-30T08:45:08.706Z 'ERROR :' 'Request timed out.'
2017-03-30T08:45:08.708Z 'DEBUG :' 'waiting' 'error'
2017-03-30T08:45:08.710Z 'ERROR :' 'Client in error state.'
{ err: 'timeout' }
```

stefanpoeter, 2017-03-30:
Some code please.

NoobTW, 2017-03-30:
```
var modbus = require('jsmodbus');

// create a modbus client
var client = modbus.client.tcp.complete({
        'host'              : '192.168.255.2',
        'port'              : 502,
        'autoReconnect'     : true,
        'reconnectTimeout'  : 1000,
        'timeout'           : 5000,
        'unitId'            : 0,
        'logEnabled': true,
        'logLevel': 'debug',
        'logTimestamp': true
    });

client.connect();

// reconnect with client.reconnect()

client.on('connect', function () {

    // make some calls

    client.readCoils(0, 13).then(function (resp) {

        // resp will look like { fc: 1, byteCount: 20, coils: [ values 0 - 13 ], payload: <Buffer> }
        console.log(resp);

    }, console.error);

});

client.on('error', function (err) {

    console.log(err);

})
```

stefanpoeter, 2017-03-30:
> 2017-03-30T08:44:57.629Z 'DEBUG :' 'Data flushed.'
2017-03-30T08:45:02.636Z 'ERROR :' 'Request timed out.'

Seems like the request is not getting handled. What server are you using?

NoobTW, 2017-03-30:
It's a real hardware, so I'm not sure what server I'm using.

For further information, below is the code that passed in jsmodbus v0.5.2:

```
var modbus = require('jsmodbus');
var client = modbus.createTCPClient(502, '192.168.255.2', function(err) {
    if (err) {
        console.log('Cannot connect to server');
        return false;
    } else {
        console.log('Connected to server');
    }
});

client.readCoils(0, 8, function(resp, err) {
    if (!err) {
        console.log(resp.coils);
    } else {
        console.log(err);
    }
});
```

stefanpoeter, 2017-03-30:
Well, thats not much. Can you start wireshark and inspect the tcp packages?

NoobTW, 2017-03-30:
I don't think I can do that because those things are pass through a switch.

stefanpoeter, 2017-03-30:
But you are executing the client on your local machine, so wireshark is no problem.

NoobTW, 2017-03-30:
Oops, you're right. Because I ran the script remotely on a Raspberry Pi.

[HERE](https://drive.google.com/open?id=0BzFytfpAXJmFVHhlcXVOaDRNU3c) is the pcap file I caught using tshark.

stefanpoeter, 2017-03-30:
The way I see it, there is no response coming from your device. Check your devices configuration, Doesn't seem like an error related to this module.

NoobTW, 2017-03-30:
That's weird the device only reply the old version jsmodbus. But thanks, I'll check if it's a device issue.

closed 2017-03-30 by NoobTW

NoobTW, 2017-03-30:
Hi.

After a few try,  I found out that change `unitId: 1` then my device will handle it.
Just let you know, and thanks for your help.