Cloud-Automation/node-modbus #131: Connect timeout feature request

klmurphy72, 2017-04-01:
It would be nice if there was an option for a connect timeout.  It would make connecting to a device that was just plugged in much faster!

thank you

stefanpoeter, 2017-04-03:
What about tcp probe? There was a issue a while back.

klmurphy72, 2017-04-03:
tcp probe?  I don't understand.  If this should mean something to me, sorry....it does not.

stefanpoeter, 2017-04-04:
Have a look at issue #36 You can use the npm package tcp-ping to probe a tcp port. This way you can check if a modbus device is online periodically without risking a connection error.

stefanpoeter, 2017-05-02:
Have a look at how the reconnect is handled in the v3.0.0-alpha branch. The socket logic is separated from the modbus client.

closed 2017-05-02 by stefanpoeter