Cloud-Automation/node-modbus #132: Error when timeout

AurelienC, 2017-04-28:
Hello,

I use this package with Meteor.

When I disconnect my PLC on the TCP network, and after some time, I get this error : 

`W20170428-15:55:04.950(2)? (STDERR) /home/meteorjs/MeteorSupervision/node_modules/stampit-event-bus/src/stampit-event-bus.js:20`
`W20170428-15:55:04.951(2)? (STDERR)             cbList[name][i].listener.apply(this, args);`
`W20170428-15:55:04.953(2)? (STDERR)                                     ^ // on the dot between listener and apply` 
`W20170428-15:55:04.953(2)? (STDERR) `
`W20170428-15:55:04.953(2)? (STDERR) TypeError: Cannot read property 'apply' of undefined`
`W20170428-15:55:04.954(2)? (STDERR)     at Object.emit (/home/meteorjs/MeteorSupervision/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:37)`
`W20170428-15:55:04.954(2)? (STDERR)     at Object.<anonymous> (/home/meteorjs/MeteorSupervision/node_modules/jsmodbus/src/modbus-tcp-client.js:88:18)`
`W20170428-15:55:04.954(2)? (STDERR)     at emitOne (events.js:77:13)`
`W20170428-15:55:04.955(2)? (STDERR)     at Socket.emit (events.js:169:7)`
`W20170428-15:55:04.957(2)? (STDERR)     at emitErrorNT (net.js:1269:8)`
`W20170428-15:55:04.958(2)? (STDERR)     at nextTickCallbackWith2Args (node.js:511:9)`
`W20170428-15:55:04.958(2)? (STDERR)     at process._tickCallback (node.js:425:17)`
`=> Exited with code: 1`

This error fall off the Meteor process.

Any idea of the reason ?

Many thanks 😄 
Aurélien

stefanpoeter, 2017-04-28:
Some code please.

AurelienC, 2017-05-02:
Hi,

Following, the called code during the disconnection of the PLC : 
`this.client.readHoldingRegisters(this.equipment.connection.offset, 30)`
`      .then((resp) => {`
`        this.readValue = resp;`
`        callback(resp);`
`      })`
`      .fail((err) => {`
`        console.log(err);`
`          Meteor.Error('modbus.timeout', this.equipment.ip + ' timeout.', 'Impossible to contact equipment.');`
`      });`

The function `readHoldingRegisters` is calling by CRON every several seconds.
I don't disconnect the PLC between two calling.

First time, I get an timeout error into the `fail`. Then, neither until the crash.

[image]

Thanks 😄

stefanpoeter, 2017-05-02:
There are some retransmission tcp packages in your log. Could be that the client tries to retransmit the package and while doing so the timeout sets in but then the retransmission was successfull. Could you put a console.log in the then part and numerate your requests?

AurelienC, 2017-05-02:
__CRON job__ _call periodiqueTask_
`new CronJob(cron, Meteor.bindEnvironment(() => {`
`        console.log('Cron calling');`
`        equipmentsMb[equipment._id].periodiqueTask();`
`      }), null, true, 'Europe/Paris'),`

__periodiqueTask__  _call getData_
`periodiqueTask() {`
`    this.getData(Meteor.bindEnvironment(() => {`
`      this.qtyCall += 1;`
`      console.log('Calling n° ' + this.qtyCall);`

__getData__ _call readHoldingRegisters_
`getData(callback) {`
`      this.client.readHoldingRegisters(this.equipment.connection.offset, 30)`
`      .then((resp) => {`
`        this.readValue = resp;`
`        callback(resp);`
`      })`
`      .fail((err) => {`
`        console.log('fail');`
`        console.log(err);`
`      });`
`  }`

__and__ 🥁 🥁 🥁 _(drum roll)_ ... 
__console.log !__

`I20170502-15:45:12.745(2)? Cron calling`
`I20170502-15:45:12.750(2)? Calling n° 1`
`I20170502-15:45:15.746(2)? Cron calling`
`I20170502-15:45:15.750(2)? Calling n° 2`
`I20170502-15:45:18.749(2)? Cron calling`
`I20170502-15:45:18.753(2)? Calling n° 3`
`I20170502-15:45:21.753(2)? Cron calling`
`I20170502-15:45:23.753(2)? fail`
`I20170502-15:45:23.754(2)? { err: 'timeout' }`
`I20170502-15:45:24.756(2)? Cron calling // 50x`

`W20170502-15:47:41.048(2)? (STDERR) /home/caau/meteorjs/sfsMeteorSupervision/node_modules/stampit-event-bus/src/stampit-event-bus.js:20`
`W20170502-15:47:41.049(2)? (STDERR)             cbList[name][i].listener.apply(this, args);`
`W20170502-15:47:41.049(2)? (STDERR)`                                     ^
`W20170502-15:47:41.050(2)? (STDERR) `
`W20170502-15:47:41.050(2)? (STDERR) TypeError: Cannot read property 'apply' of undefined`
`W20170502-15:47:41.051(2)? (STDERR)     at Object.emit (/home/caau/meteorjs/sfsMeteorSupervision/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:37)`
`W20170502-15:47:41.051(2)? (STDERR)     at Object.<anonymous> (/home/caau/meteorjs/sfsMeteorSupervision/node_modules/jsmodbus/src/modbus-tcp-client.js:88:18)`
`W20170502-15:47:41.051(2)? (STDERR)     at emitOne (events.js:77:13)`
`W20170502-15:47:41.052(2)? (STDERR)     at Socket.emit (events.js:169:7)`
`W20170502-15:47:41.052(2)? (STDERR)     at emitErrorNT (net.js:1269:8)`
`W20170502-15:47:41.052(2)? (STDERR)     at nextTickCallbackWith2Args (node.js:511:9)`
`W20170502-15:47:41.052(2)? (STDERR)     at process._tickCallback (node.js:425:17)`
`=> Exited with code: 1`

__When connected is down up__ : (all calls are performed in one time)
`I20170502-15:48:48.202(2)? Cron calling`
`I20170502-15:48:50.541(2)? Calling n° 1`
`I20170502-15:48:50.566(2)? Calling n° 2`
`I20170502-15:48:50.591(2)? Calling n° 3`
`I20170502-15:48:50.601(2)? Calling n° 4`
`I20170502-15:48:50.660(2)? Calling n° 5`
`I20170502-15:48:50.675(2)? Calling n° 6`
`I20170502-15:48:50.712(2)? Calling n° 7`
`I20170502-15:48:50.761(2)? Calling n° 8`
`I20170502-15:48:50.803(2)? Calling n° 9`
`I20170502-15:48:50.837(2)? Calling n° 10`
`I20170502-15:48:50.875(2)? Calling n° 11`
`I20170502-15:48:50.908(2)? Calling n° 12`
`I20170502-15:48:50.932(2)? Calling n° 13`
`I20170502-15:48:50.964(2)? Calling n° 14`
`I20170502-15:48:50.998(2)? Calling n° 15`
`I20170502-15:48:51.032(2)? Calling n° 16`
`I20170502-15:48:51.057(2)? Calling n° 17`
`I20170502-15:48:51.095(2)? Calling n° 18`
`I20170502-15:48:51.128(2)? Calling n° 19`
`I20170502-15:48:51.155(2)? Calling n° 20`
`I20170502-15:48:51.183(2)? Calling n° 21`

Maybe I not use correctly the library ?

AurelienC, 2017-05-02:
With `logLvel: debug` :

`I20170502-16:06:18.898(2)? Cron calling`
`I20170502-16:06:18.898(2)? DEBUG : Starting read holding registers request.`
`I20170502-16:06:18.899(2)? DEBUG : Trying to flush data.`
`I20170502-16:06:18.899(2)? DEBUG : ready waiting`
`I20170502-16:06:18.899(2)? DEBUG : Sending pdu to the socket.`
`I20170502-16:06:18.899(2)? DEBUG : Data flushed.`
`I20170502-16:06:20.900(2)? ERROR : Request timed out.`
`I20170502-16:06:20.902(2)? DEBUG : waiting error`
`I20170502-16:06:20.903(2)? ERROR : Client in error state.`
`I20170502-16:06:20.904(2)? fail`
`I20170502-16:06:20.904(2)? { err: 'timeout' }`
`I20170502-16:06:20.904(2)? DEBUG : Socket closed with error false`
`I20170502-16:06:20.905(2)? DEBUG : error closed`
`I20170502-16:06:20.905(2)? DEBUG : Clearing timeout of the current request.`
`I20170502-16:06:20.905(2)? DEBUG : Cleaning up request fifo.`
`I20170502-16:06:21.899(2)? Cron calling`
`I20170502-16:06:21.900(2)? DEBUG : Starting read holding registers request.`
`I20170502-16:06:24.902(2)? Cron calling`
`I20170502-16:06:24.902(2)? DEBUG : Starting read holding registers request.`

And when the crash become : 
`I20170502-16:08:36.041(2)? Cron calling`
`I20170502-16:08:36.042(2)? DEBUG : Starting read holding registers request.`
`I20170502-16:08:38.252(2)? ERROR : Socket Error { [Error: connect ETIMEDOUT 10.33.6.19:502]`
`I20170502-16:08:38.252(2)?   code: 'ETIMEDOUT',`
`I20170502-16:08:38.253(2)?   errno: 'ETIMEDOUT',`
`I20170502-16:08:38.253(2)?   syscall: 'connect',`
`I20170502-16:08:38.255(2)?   address: '10.33.6.19',`
`I20170502-16:08:38.255(2)?   port: 502 }`
`I20170502-16:08:38.256(2)? DEBUG : connect error`
`I20170502-16:08:38.256(2)? ERROR : Client in error state.`
`W20170502-16:08:38.258(2)? (STDERR) /home/caau/meteorjs/sfsMeteorSupervision/node_modules/stampit-event-bus/src/stampit-event-bus.js:20`
`W20170502-16:08:38.258(2)? (STDERR)             cbList[name][i].listener.apply(this, args);`
`W20170502-16:08:38.259(2)? (STDERR)                                     ^`
`W20170502-16:08:38.259(2)? (STDERR) `
`W20170502-16:08:38.259(2)? (STDERR) TypeError: Cannot read property 'apply' of undefined`
`W20170502-16:08:38.259(2)? (STDERR)     at Object.emit (/home/caau/meteorjs/sfsMeteorSupervision/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:37)`
`W20170502-16:08:38.260(2)? (STDERR)     at Object.<anonymous> (/home/caau/meteorjs/sfsMeteorSupervision/node_modules/jsmodbus/src/modbus-tcp-client.js:88:18)`
`W20170502-16:08:38.260(2)? (STDERR)     at emitOne (events.js:77:13)`
`W20170502-16:08:38.260(2)? (STDERR)     at Socket.emit (events.js:169:7)`
`W20170502-16:08:38.260(2)? (STDERR)     at emitErrorNT (net.js:1269:8)`
`W20170502-16:08:38.261(2)? (STDERR)     at nextTickCallbackWith2Args (node.js:511:9)`
`W20170502-16:08:38.261(2)? (STDERR)     at process._tickCallback (node.js:425:17)`
`=> Exited with code: 1`

AurelienC, 2017-05-02:
@stefanpoeter May be this error is caused by stampit ? This error may be resolved with #133 ? What did you think about ?

AurelienC, 2017-05-03:
I just updated with 2.1.0 version and no new error about Socket error.

Thanks for your package @stefanpoeter 👍

closed 2017-05-03 by AurelienC