Cloud-Automation/node-modbus #133: Version 3 Announcement

stefanpoeter, 2017-05-02:
Hi all,

i've reimplemented this module to address some issues that were coming up from time to time and to get rid of some dependencies. You can have a look at the code on the v3.0.0-alpha branch. The code follows strictly the composition over inheritance pattern and doesn't use stampit anymore. Have a look in the Readme on that branch for some more informations and give me some feedback. The Server isn't yet implemented.

Thanks,
Stefan

stefanpoeter, 2017-05-22:
I've made some progress here. The TCP and RTU Client are already complete, unfortunatly there is an issue with the serialport implementation and ubuntu where I cannot send or receive information. I compared the rtu output for all requests, it looks fine, I just couldn't test it. Is anyone out there who wants to try the rtu client?

koalex, 2017-08-16:
Is it planned to release a stable version?

stefanpoeter, 2017-08-16:
It is supposed to be a stable version. I am looking for people to test it. Would be happy if you do. :-)

profelec, 2017-08-16:
But the Server isn't yet implemented.

profelec, 2017-09-15:
Hi,
Is it planned to release the server?
thanks for your job ;-)

stefanpoeter, 2017-09-18:
Well yes, it comes with a server and a client. There is still some work that needs to be done, any help yould be appreciated.

profelec, 2017-10-02:
I do not understand this line, in jsmodbus/src/tcp-server-request-handler.js  : 
this._buffer = Buffer.compose([this._buffer, data])
"compose" does not exist

stefanpoeter, 2017-10-02:
This should be `Buffer.concat([this._buffer, data])`. This code is probably untested yet.

profelec, 2017-10-02:
the server does not crash anymore, but the client shows me this:
```
Automate connected.
Unhandled rejection (<{"err":"timeout"}>, no stack trace)

Client closed, stopping interval.
Automate connected.
Unhandled rejection (<{"err":"timeout"}>, no stack trace)

Client closed, stopping interval.
```

stefanpoeter, 2017-10-02:
The tcp server response handler only handles read coil requests by now. Have a look at the tcp-server-repsonse-handler.js file.

AnthonyMujic, 2017-12-28:
I find this version really intuitive - great work.
I don't have a serial port on my computer so I don't think I can help testing the RTU Client, but I would love to work on the TCP Server. Is it as simple as making pull requests, or is there someone I should get in contact with first?

stefanpoeter, 2017-12-28:
Great to hear, you can simply make pull requests.

closed 2018-02-21 by stefanpoeter