Cloud-Automation/node-modbus #134: Maximum limit of clients?

tgirotto, 2017-05-13:
Hi, I am using jsmodbus to read from different hosts in modbus tcp. Consequently, I need multiple clients in the same node application. However, it seems that I can only have a maximum of two clients. Can you confirm that? Is there any other way I could read from multiple servers?

Thank you!

stefanpoeter, 2017-05-13:
Not that I am aware of. You can have as many clients in one application as you like.

tgirotto, 2017-05-14:
@stefanpoeter I have tested the issue with a basic interval loop and a physical PLC controller (Eliwell Advance).

    var modbus = require('jsmodbus');

    var counter = 0;
    var clients = [];

    function launchClient(input, cb) {
        var client = modbus.client.tcp.complete({
            'host'              : input.host,
            'port'              : input.port,
            'autoReconnect'     : true,
            'reconnectTimeout'  : 1000,
            'timeout'           : 5000,
            'unitId'            : input.slaveId
        });

        clients.push(client);
        client.connect();
        client.on("connect", function() {
            cb();
        })
    }

    setInterval(function() {
	    counter += 1;
	     var i = {
                    name: counter,
                    host: '10.0.0.100',
                    port: 502,
                    address: 8959,
                    slaveId: 255,
                    dataType: 'int',
                    interval: 5
                };

    	    launchClient(i, function() {
		    console.log('connected ' + i.name + ' to PLC')
	    });
    }, 5000);`

 It looks like the first three clients connect just fine, but after the third one is created, the 'connect' event that is supposed to be emitted by the Socket object stops triggering the callback.

stefanpoeter, 2017-05-14:
What does the documentation from your plc say. Sometimes they limit the amount of client connections.

tgirotto, 2017-05-14:
mm the docs don't say anything about it, but I have made two further experiments:

1) I have connected two PLC controllers at the same time, and tried to add multiple clients to both. As a result, I could create more clients, but after the third client on each, the requests started timing out;
2) I have run another loop that creates bare sockets to connect to the controllers. Again, the max limit was three. 

Conclusion: it's likely to be a limitation of the PLC controller.

closed 2017-05-14 by stefanpoeter