Cloud-Automation/node-modbus #137: Error with socket (undefined) when calling .close()

AurelienC, 2017-06-09:
Hi,

I'm using this package with Meteor.JS and I have an throwing exception.

My client is created and the state is `init`.

I'm just calling `myEquipment.close();`

```
W20170609-14:20:11.722(2)? (STDERR) TypeError: Cannot read property 'end' of undefined
W20170609-14:20:11.723(2)? (STDERR)     at Object.close (/home/meteorjs/MeteorSupervision/node_modules/jsmodbus/src/modbus-tcp-client.js:173:13)
```

The [line](https://github.com/Cloud-Automation/node-modbus/blob/167281a95105151b2b03d5b5ac94a229b31d9b02/src/modbus-tcp-client.js#L173) into the package file.

Maybe the socket has been not created, if I close before ? But, it will be good if a test will made before calling `socket.end()` into `close` function ? What did you think about ?

stefanpoeter, 2017-06-09:
Hey @AurelienC,

i am currently on vacation and can not reproduce your error. Can you post some code. That might help.

AurelienC, 2017-06-13:
Hello,

I have not much time now for work on this fault. Enjoy your vacation ;-) 🏖

My code is really simple. Maybe, sometime, I try to stop connection before to be connected.
```

// Creation (into method)
this.client = modbus.client.tcp.complete({
      host: this.equipment.connection.ip,
      port: this.equipment.connection.port,
      autoReconnect: true,
      // logEnabled: true,
      // logLevel: 'debug',
      reconnectTimeout: 10000,
      timeout: 2000,
      unitId: this.equipment.connection.id,
    });

// Connection (into method)
  this.client.connect();

// Close (into method)
this.client.close()
```

stefanpoeter, 2017-06-19:
You should wait with the `this.client.close` call after the connection is established. The client emits a `online` signal when connected.

closed 2017-06-26 by stefanpoeter

AurelienC, 2017-06-26:
Sorry for delay but my time is very restricted on this project.

I have make a better code and I no longer have this error now.

If I have more time, I will try to propose a pull request with a solution.

Thanks for your package @stefanpoeter  🖐