Cloud-Automation/node-modbus #138: Modbus Serial errors

DzmitryT, 2017-06-13:
Hello, I'm trying to use your library for testing of a couple of devices connected to RPi via FTDI-based USB-serial adapter and it seems there some errors in examples provided. Also, there is critical bug in the modbus-serial-client.js.
`    var onSend = function (pdu) {
      var crc = 0

      var base = Buffer.allocUnsafe(1)
      base.writeUInt8(1)
      var buf = Buffer.concat([base, pdu])

      for (var i = 0; i < buf.length; i += 1) {
        crc = (buf.readUInt8(i) + crc) % 0xFFFF
      }

      var crc16 = crc.crc16modbus(buf)`

Seems, this code should be replaced with something like that:
`    var onSend = function (pdu) {
      var tmp = 0

      var base = Buffer.allocUnsafe(1)
      base.writeUInt8(1)
      var buf = Buffer.concat([base, pdu])

      for (var i = 0; i < buf.length; i += 1) {
        tmp = (buf.readUInt8(i) + tmp) % 0xFFFF
      }

      var crc16 = crc.crc16modbus(buf)`

Because an error occurred when function **crc16modbus** called on the number, not as method of the **crc** object:
> TypeError: crc.crc16modbus is not a function

Regards, Dzmitry.

stefanpoeter, 2017-06-19:
Hey @DzmitryT what version are u using?

DzmitryT, 2017-06-19:
According to **package.json**, there is a latest version

>   "_from": "jsmodbus@latest",
  "_id": "jsmodbus@2.1.0",

stefanpoeter, 2017-06-26:
I'll make an update. Thanks for the tip.

stefanpoeter, 2017-06-26:
Tried to fix it with issue#138. I made a pull request for it. @DzmitryT  can you check if it is gone?

DzmitryT, 2017-06-26:
Thx, I'll try to check your update some hours later or tomorrow.

DzmitryT, 2017-06-28:
@stefanpoeter, apologize for the late reply. Updated code seems to be working.

closed 2017-06-29 by stefanpoeter