Cloud-Automation/node-modbus #141: Cant call close multiple times

jbockerstette, 2017-07-19:
If I call close more than once, it throws TypeError. I personally think it would be better to allow you to call close multiple times on the same modbus client without throwing the TypeError. This is not a show stopper for me at all. Your code is great!

stefanpoeter, 2017-07-21:
Thanks for the hint. I'll make an update.

stefanpoeter, 2017-07-28:
Can you confirm that pull request #142 fixes this?

jbockerstette, 2017-07-28:
I verified that now I can call close multiple times without throwing a TypeError. What was happening is that when you call close, socket.end() gets called. If you call it again, then socket.end() would get called again but on the second call, socket === null and then it would throw. So yes, this is fixed.

Thanks.

closed 2017-08-01 by stefanpoeter