Cloud-Automation/node-modbus #144: Real Number Handling

awadekaustubh, 2017-08-22:
Hi 

First of all thanks for a wonderful utility code for connecting modbus devices. Life has become much easier with this. 

While testing it, on modbus plc we came across a particular use case, the data coming on PLC is in REAL data type but we are not able to handle it in opc server. It is very crucial for us to be able to read REAL data type values as it is extensively used in industries. It is very difficult to implement this solution if it is not able to intercept real values.

Can you suggest a workaround or update the code to adapt to real values.

stefanpoeter, 2017-08-22:
Hey there,
have a look at issue #114. It is already handled in there.

stefanpoeter, 2017-08-28:
I'll close this since @blhough answered your questing in issue #114.

closed 2017-08-28 by stefanpoeter