Cloud-Automation/node-modbus #145: ReadDiscrete use getInput

davidonet, 2017-09-05:
Why does ReadDiscrete use Input Buffer it's a bit confusing ?

https://github.com/Cloud-Automation/node-modbus/blob/321f71f15ad7f6516665bd7cc755a71232fba1b8/src/handler/server/ReadDiscreteInputs.js#L38

I can't find getDiscrete method.

stefanpoeter, 2017-09-05:
It is indeed confusing. I'll always used it this way, there was an issue some time back that adressed this issue too. I'll try to avoid this in the v3 release.

davidonet, 2017-09-08:
How can I avoid collision between a ReadInput(0x00) and a ReadDiscrete(0x00) ?

stefanpoeter, 2017-09-08:
If you want to do it right then I would make a pull request and replace the input buffer with a new discrete buffer. That way you are fine.

davidonet, 2017-09-08:
Yes please :-)

stefanpoeter, 2017-11-01:
Sorry for the long delay. Can you confirm that this is working the way you wanted it?

davidonet, 2017-11-06:
I've seen a pull request that seems to fix it : https://github.com/Cloud-Automation/node-modbus/pull/154
But it doesn't seem to be integrated in the current master branch : https://github.com/Cloud-Automation/node-modbus/blob/1cc5a9088a904c60470ac865c4ab2d24445cf655/src/handler/server/ReadDiscreteInputs.js#L38
There is still a getInput buffer in the ReadDiscreteInput.

stefanpoeter, 2017-11-06:
Yes, you need to confirm that the pull request fixed the issue before I merge it into the master branch.

davidonet, 2017-11-07:
I don't know how to test it, I tried to fork the project but I can't merge the pull request in my fork.

davidonet, 2017-11-07:
I figure how it can be done locally : 
https://help.github.com/articles/checking-out-pull-requests-locally/

davidonet, 2017-11-07:
Yes it works Input buffer and Discrete buffer are distinct.
tested with 
```javascript 
// in SimpleServer
this.getInput().writeUInt16BE(0xff,0)
this.getDiscrete().writeUInt8(0xaa,0)
```
and client
``` shell
$ node ReadDiscreteInput.js localhost 8888 0 1
$ node ReadInputRegister.js localhost 8888 0 1
```

stefanpoeter, 2017-11-07:
Can you also copy the output from the  command line. Thanks.

davidonet, 2017-11-07:
done

stefanpoeter, 2017-11-07:
Thanks, I'll merge it soon then.

closed 2017-11-07 by stefanpoeter