Cloud-Automation/node-modbus #146: error on handleErrorPDU() after timeout

JasLin, 2017-09-08:
I got an error after the client idle for several minutes, may be modbus device return error PDU, I have't capture the tcp package yet (I will do that later) . I am think about that should handleErrorPDU try catch ```unexceptional errors```, and made it more robust? 

```
Index out of range","type":"Error","stack":"RangeError: Index out of range\n    at checkOffset (buffer.js:970:11)\n    at Buffer.readUInt8 (buffer.js:1008:5)\n    at handleErrorPDU (/home/apps/hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:76:27)\n    at Object.<anonymous> (/home/apps/hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:117:11)\n    at Object.emit (/home/apps/hardware-controller/node_modules/_stampit-event-bus@0.1.1@stampit-event-bus/src/stampit-event-bus.js:20:38)\n    at Object.<anonymous> (/home/apps/hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-tcp-client.js:112:14)\n    at emitOne (events.js:115:13)\n    at Socket.emit (events.js:210:7)\n    at addChunk (_stream_readable.js:266:12)\n    at readableAddChunk (_stream_readable.js:253:11)\n    at Socket.Readable.push (_stream_readable.js:211:10)\n    at TCP.onread (net.js:585:20)","v":1}
```

```
exports.writeSingleCoil = async function(address, value) {
    return await client.writeSingleCoil(address, value)
    .catch(error => {
        log.error(error)
        return null
    }) 
}
```

stefanpoeter, 2017-09-08:
Can you give me a code example to reproduce the error.

JasLin, 2017-09-12:
@stefanpoeter thanks for reply.

I have found the cause of this issue, the device we got have an custom heat beat (different between  tcp keep-alive package). since jsmodbus can not reconginze this custom package, then it close the socket. 

* the custom heat beat:

[image]

* standard tcp keep-alive package 

[image]

JasLin, 2017-09-12:
so , I am think about that , should we try catch those  case(unrecognizable package), and  emit en ```error``` event instead of close the socket?

JasLin, 2017-09-12:
more details here:

1. if we trying to writeCoils after socket closed ,  it raise ```ECONNRESET```
···
DEBUG : Trying to flush data.
DEBUG : ready waiting
DEBUG : Sending pdu to the socket.
DEBUG : Data flushed.
ERROR : Socket Error { Error: read ECONNRESET
    at _errnoException (util.js:1041:11)
    at TCP.onread (net.js:606:25) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
DEBUG : waiting error
ERROR : Client in error state.
DEBUG : Socket closed with error true
DEBUG : error closed
DEBUG : Clearing timeout of the current request.
DEBUG : Cleaning up request fifo.

···

2. client auto reconnected to device (modbus server)

```
{"pid":13399,"hostname":"cubieboard2","name":"relay-controller","level":20,"time":1505198502611,"msg":"init().onConnect connected to relay server: 192.168.3.233 : 10000","v":1}
```

3.  device  (modbus server) send an custom heat beat package , and this package lead the Error.

```
DEBUG : received data
{"pid":13399,"hostname":"cubieboard2","name":"facade","level":50,"time":1505198548092,"msg":"on uncaughtException","v":1}
{"pid":13399,"hostname":"cubieboard2","name":"facade","level":50,"time":1505198548094,"msg":"Index out of range","type":"Error","stack":"RangeError: Index out of range\n    at checkOffset (buffer.js:970:11)\n    at Buffer.readUInt8 (buffer.js:1008:5)\n    at handleErrorPDU (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:76:27)\n    at Object.<anonymous> (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:117:11)\n    at Object.emit (/home/apps/selfservice-store-hardware-controller/node_modules/_stampit-event-bus@0.1.1@stampit-event-bus/src/stampit-event-bus.js:20:38)\n    at Object.<anonymous> (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-tcp-client.js:112:14)\n    at emitOne (events.js:115:13)\n    at Socket.emit (events.js:210:7)\n    at addChunk (_stream_readable.js:266:12)\n    at readableAddChunk (_stream_readable.js:253:11)\n    at Socket.Readable.push (_stream_readable.js:211:10)\n    at TCP.onread (net.js:585:20)","v":1}
DEBUG : received data
DEBUG : PDU extracted
DEBUG : received data
{"pid":13399,"hostname":"cubieboard2","name":"facade","level":50,"time":1505198615808,"msg":"on uncaughtException","v":1}
{"pid":13399,"hostname":"cubieboard2","name":"facade","level":50,"time":1505198615809,"msg":"Index out of range","type":"Error","stack":"RangeError: Index out of range\n    at checkOffset (buffer.js:970:11)\n    at Buffer.readUInt8 (buffer.js:1008:5)\n    at handleErrorPDU (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:76:27)\n    at Object.<anonymous> (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:117:11)\n    at Object.emit (/home/apps/selfservice-store-hardware-controller/node_modules/_stampit-event-bus@0.1.1@stampit-event-bus/src/stampit-event-bus.js:20:38)\n    at Object.<anonymous> (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-tcp-client.js:112:14)\n    at emitOne (events.js:115:13)\n    at Socket.emit (events.js:210:7)\n    at addChunk (_stream_readable.js:266:12)\n    at readableAddChunk (_stream_readable.js:253:11)\n    at Socket.Readable.push (_stream_readable.js:211:10)\n    at TCP.onread (net.js:585:20)","v":1}
DEBUG : received data
DEBUG : PDU extracted
DEBUG : received data
{"pid":13399,"hostname":"cubieboard2","name":"facade","level":50,"time":1505198680900,"msg":"on uncaughtException","v":1}
{"pid":13399,"hostname":"cubieboard2","name":"facade","level":50,"time":1505198680901,"msg":"Index out of range","type":"Error","stack":"RangeError: Index out of range\n    at checkOffset (buffer.js:970:11)\n    at Buffer.readUInt8 (buffer.js:1008:5)\n    at handleErrorPDU (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:76:27)\n    at Object.<anonymous> (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-client-core.js:117:11)\n    at Object.emit (/home/apps/selfservice-store-hardware-controller/node_modules/_stampit-event-bus@0.1.1@stampit-event-bus/src/stampit-event-bus.js:20:38)\n    at Object.<anonymous> (/home/apps/selfservice-store-hardware-controller/node_modules/_jsmodbus@2.1.2@jsmodbus/src/modbus-tcp-client.js:112:14)\n    at emitOne (events.js:115:13)\n    at Socket.emit (events.js:210:7)\n    at addChunk (_stream_readable.js:266:12)\n    at readableAddChunk (_stream_readable.js:253:11)\n    at Socket.Readable.push (_stream_readable.js:211:10)\n    at TCP.onread (net.js:585:20)","v":1}
```

stefanpoeter, 2017-09-12:
Thanks for your input. I say that data regarding the transport layer (keep-alive) for example are not supposed to be delivered to the application layer. If this data is transmitted to the application layer and is supposed to be ment for the transportation layer then something underneath is wrong.

When it comes to handling wrong packages I say it is difficult to do. In this case there is not enough data transmitted to cover a whole modbus message,  so we have to wait for more data (which won't arrive). If we get wrong data that still fits into a modbus message frame than some error will be shot.

JasLin, 2017-09-12:
yes, keep-alive package is not the cause of the error, because it's not delivered to application layer. 

I our case , there are some custom heat beat package from the third-party device. those package lead the error.  and it is enough data transmitted, but it's not a standard modbus message.

stefanpoeter, 2017-09-12:
You probably mean a heartbeat package? This is, to my knowledge not part of the modbus specification and shouldn't be handled in this module.

JasLin, 2017-09-12:
:). yes , heartbeat (sorry for the wrong words ```heat beat``` ) , and I knew heartbeat it not a port of modbus spec . I mean , I don't want to implements  heartbeat in my application. we had bought some modbus devices , those devices implements heartbeat in themselves, and they sent those heartbeat package to jsmodbus, those heartbeat packages drive jsmodbus into error.  this is exactly what the problem I had met currently . as u can see in the debug log and screenshot.

stefanpoeter, 2017-09-12:
You can always create a fork of this project and change it to your needs. Since these heartbeat packages are not part of the modbus specification, changes will not go into the main source.

JasLin, 2017-09-12:
ok,  I will try to catch those non-modbus package. 

thanks for you help.

closed 2017-09-12 by stefanpoeter