Cloud-Automation/node-modbus #153: Accessing the resp components in a key:value manner

sergiupienar, 2017-11-01:
What methods are available on the resp object, e.g. is there any way of getting the value in a nice way ?

{ fc: 3,
   byteCount: 20,
   payload: <Buffer 00 15 00 64 00 5a 00 64 00 c8 00 64 00 c8 00 00 00
00 00 00>,
   register: [ 21, 100, 90, 100, 200, 100, 200, 0, 0, 0 ] }

Are there any getters for fc, byteCount, etc ?

stefanpoeter, 2017-11-01:
why do you want to have getters?

sergiupienar, 2017-11-01:
When dealing with the response that comes after a read request (be it on coils, registers, etc) it would be handy to fetch the payload in a nicer manner - i.e. resp.getPayload(), resp.getRegister().

stefanpoeter, 2017-11-01:
i disagree, the response is a simple json object, the easiest form to store complex data in javascript. I don't see any need to prototype the response with getters and setters. If you were using es5-class-getter the syntax would be exactly the same (resp.payload and resp.register)

closed 2017-11-01 by sergiupienar