Cloud-Automation/node-modbus #155: Error while connecting to modbus device with jsmodebus and async.

georgenici, 2017-11-08:
Hi!
I created the program that makes 2 parallel requests to modbus devices using jsmodbus (node.js driver) and asyncjs every 3 seconds. I get an error after some time (random periods of time) during the first run and if i try to run it again i immediately get the same error.

Error:
[ { value:
    { fc: 3,
       byteCount: 20,
       payload: <Buffer 00 6f 00 0b 00 16 00 21 00 2c 00 37 00 42 00 4d 00 58 00 63>,
       register: [Array] } },
  { value:
     { fc: 3,
       byteCount: 20,
       payload: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>,
       register: [Array] } } ]
[ { value:
     { fc: 3,
       byteCount: 20,
       payload: <Buffer 00 6f 00 0b 00 16 00 21 00 2c 00 37 00 42 00 4d 00 58 00 63>,
       register: [Array] } },
  { value:
     { fc: 3,
       byteCount: 20,
       payload: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>,
       register: [Array] } } ]
Unhandled rejection (<{"err":"timeout"}>, no stack trace)
Unhandled rejection (<{"err":"timeout"}>, no stack trace)
Unhandled rejection (<{"err":"timeout"}>, no stack trace)
Unhandled rejection (<{"err":"timeout"}>, no stack trace)

My code:
const modbus = require('jsmodbus');
    const async = require('async');
    setInterval(()=>{
        async.parallel([
            async.reflect(function(callback) {
                var client = modbus.client.tcp.complete({ 
                    'host'              : '192.168.56.112', 
                    'port'              : 502,
                    'autoReconnect'     : false,
            'reconnectTimeout'  : 1000,
            'timeout'           : 1000,
            'unitId'            : 0
        }).connect();
                   client.on('connect', function () {
                client.readHoldingRegisters(0, 10).then(function (resp) {
              return callback(null, resp); 
           }).finally(function () {
            client.close();
          });
        });
        client.on('error', function (err) {
        return callback('Connection error1');
        });
    }),
async.reflect(function(callback) {
var client = modbus.client.tcp.complete({ 
    'host'              : '192.168.56.112', 
    'port'              : 503,
    'autoReconnect'     : false,
    'reconnectTimeout'  : 1000,
    'timeout'           : 1000,
    'unitId'            : 1
}).connect();  
           client.on('connect', function () {
        client.readHoldingRegisters(0, 10).then(function (resp) {
      return callback(null, resp); 
   }).finally(function () {
    client.close();
  });
});  
client.on('error', function (err) {
return callback('Connection error2');
});})],
function(err, results) {
    console.log(results);
}); 
},3000);

Thank you.

stefanpoeter, 2017-11-08:
You close the client Connection after the first request on each client.

georgenici, 2017-11-08:
We need to close connection on each request and create new connection each time we read data. But we can't close and connections are duplicated. This causes reaching the max number of connections.

stefanpoeter, 2017-11-08:
Can you format your code a bit. It is really hard to read.

georgenici, 2017-11-08:
Sorry. now, i guess. it's better.

stefanpoeter, 2017-11-08:
I have no idea what the async module is doing. Can you try stripping away everything not related to jsmodbus and just connect two clients, make the requests and close the client and shutdown the application. Shouldn't be that hard.

georgenici, 2017-11-09:
We have several modbus devices and i'd like to request data from each of them simultaneously and then agregate all data into one array. What do you recommend in this case?

georgenici, 2017-11-09:
Thank you for your help and your product. The problem was with the emulator. On our PLC devices everything works great.
Thanks a lot, great job.
P.S. are you planning to create a bacnet/ip driver? )

closed 2017-11-09 by stefanpoeter